ruabmbua/hidapi-rs #5: Segfault on HidApi::new() when no HID devices are found.

kitling, 2016-10-14:
If I am correct, something isn't checking if it's NULL in case there's no devices, so when it tries to move stuff from the non-existant CFFI struct into a native struct... it, well, segfaults.

If I understand correctly, all that needs to be done is add a NULL check to get_hid_device_info_vector, or restructure the loop slightly.

I may try to send a PR tomorrow, but first I must sleep.

ruabmbua, 2016-10-14:
"NULL in the case of failure." says the hidapi manual. Their documentation is also slightly wrong, because no devices is indeed a possibility, but might be rare.

The rust wrapper should  check the case, but we have to decide what we want to return. I think changing the return signature to a Result might be the right choice if we want to interpret it exactly like the hidapi doc.

kitling, 2016-10-15:
I'm leaning towards an Option, given that we have no way to see if there are just none or if there was an error. That said, the final user will end up knowing if it's an error or not, given that there's a `printf("Can't create udev\n");` inside `hid_enumerate()`. >_>

KenanSulayman, 2017-06-04:
Let's be rusty about this: if `hid_enumerate()` returns NULL, we should return `None`. Result is a boolean operation -- either it works or it fails.

But if we're pragmatic, we should return an empty Iterator. That way, if one really needs to check for the length of the devices instead of matching a particular device, `api.devices().iter().collect::<Vec<&hidapi::HidDeviceInfo>>().len()` would do the job.

r8d8, 2017-07-21:
Facing this issue on a laptop, 
`hidapi` detects only one device, if I unplug it -> SeqFault.
Are there any work arounds?

ruabmbua, 2017-07-21:
This should probably be fixed in the library. A workaround would be adding an extern "C" hid_enumerate() in your application. (You should look for the correct function prototype).

Then you can call it and if it returns a NULL pointer you should omit calling HidApi::new() or refresh_devices.

ruabmbua, 2017-07-21:
I will create a PR today after work which will fix this issue.

Osspial, 2017-07-21:
#9 closes this.

closed 2017-07-21 by Osspial