ruabmbua/hidapi-rs #7: LGPL License is incompatible with Apache License

r8d8, 2017-07-16:
Hi hidapi-rs author, according to Apache website (http://www.apache.org/legal/3party.html#criteriaandcategories) , LGPL License in incompatible with Apache License, so it prevent us from using hidapi-rs in an open source project with Apache License. This make hidapi-rs not so free to use .

Is it possible to license hidapi-rs under Apache License or MIT ?

Osspial, 2017-07-16:
I wouldn't be opposed to doing this, but changing the license would need to be cleared with all of the other contributors. I'll open an issue to change this to MIT.

ruabmbua, 2017-07-16:
Also hidapi-rs uses hidapi from signal11 which has its own licenses (none of them are MIT or Apache).

Osspial, 2017-07-16:
It looks like it's available under a BSD-style license, which is compatible with both the MIT license and GPL-family licenses. From what I've seen MIT software can depend on BSD software (correct me if I'm wrong), so that shouldn't be an issue.

ruabmbua, 2017-07-16:
Okay, when that is not an issue, I as a contributor agree to relicensing it either under Apache or MIT license.

Osspial, 2017-07-16:
Closing in favor of #8.

closed 2017-07-16 by Osspial

r8d8, 2017-07-17:
Thanks, mates