ruabmbua/hidapi-rs #8: Relicense under MIT license

Osspial, 2017-07-16:
Currently, hidapi-rs is available under the LGPL license, which is unfortunately incompatible with the Apache license. Relicensing under the MIT license, using the BSD-style license from the upstream hidapi library, would allow such projects to use this crate. @ruabmbua consented in the original issue for this (#7), so right now this needs the consent of @newpavlov and @dyerti. If either of you don't give consent, please comment anyway so that your contributions may be rewritten.

Thanks!

newpavlov, 2017-07-16:
I think it's a good idea and I am giving my consent.

dyerti, 2017-07-17:
Cool with me. I consent.

r8d8, 2017-07-17:
So, we good to go with a new licence?

Osspial, 2017-07-17:
Yup! MIT has been pushed to crates.io.

closed 2017-07-17 by Osspial