ruabmbua/hidapi-rs #11: Segmentation fault when calling HidApi::new()

craigzour, 2017-11-07:
Hello,

For the second time in a month I am getting a `[1]    47775 segmentation fault` error as soon as I initialize the API `HidApi::new()` and I have no idea why it does that since it has always been working in the past. The error disappears after I restart my machine but eventually comes back at some point.

I know I am not bringing a lot of information and that it is not going to help you that much but I wanted to report the problem in case there is something you want me to try while I am having the issue.

ruabmbua, 2018-08-11:
Are you still experiencing this issue in the newest version of the library?

Also, could you please give me some detail about the platform you are targeting, and a code example?

Sorry for the very late reply (I took over crate maintenance a week ago).

ruabmbua, 2018-11-27:
I am closing this issue now. @craigzour  if you are still interested, please open a new ticket.

closed 2018-11-27 by ruabmbua