ruabmbua/hidapi-rs #20: Status of this crate?

Geobert, 2018-07-20:
Hi,

I was wondering about the state of this crate as nothing happened for a year now.

Is it deprecated? Any alternative?

Osspial, 2018-08-01:
I forked this crate from the original by @ruabmbua because I wanted to make some changes to fit my usecase and publish the changes to crates.io. However I was inexperienced with the open-source process, the original didn't seem to have much activity, and didn't want to wait for a PR to get merged/pushed to crates.io.

What I somehow didn't expect was that people would find issues with my code and then make changes to them! I lost the motivation to work on this crate pretty quickly after putting it on crates.io and (ironically enough) have left people waiting for PRs to get merged. That being said I'd be up for transferring control of this back to @ruabmbua and giving him control over the `crates.io` crate, as he seems to still be involved with its development.

ruabmbua, 2018-08-03:
Hi @Geobert , I took over the crate now.

It will get a few updates & fixes, and a new mayor version.

closed 2018-08-03 by ruabmbua