ruabmbua/hidapi-rs #23: Transfer crate ownership

ruabmbua, 2018-08-03:
Hi @Osspial.

Could you transfer ownership, or add me as a owner of the hidapi crate like you suggested in #20 ?
I am currently cherry picking all from my perspective usable additions / fixes to the crate from various forks.

The source code is available at https://github.com/ruabmbua/hidapi-rs.

Osspial, 2018-08-03:
I've invited you as an owner on crates.io and as a collaborator in this repository. It doesn't look like I can transfer ownership to you while you have a fork active, but if you delete the fork I should be able to do the transfer and then you can re-push your changes.

Sorry it took so long for this to happen!

ruabmbua, 2018-08-03:
Thanks!

Dumb question: How do I view the crates.io invite? I did not get anything.

ruabmbua, 2018-08-03:
Never mind, I found it.

ruabmbua, 2018-08-03:
Can you try to transfer repo ownership again? I deleted my fork.

Osspial, 2018-08-03:
The repo is under your username now, so it looks like it worked! Thanks for taking over.

closed 2018-08-03 by Osspial

ruabmbua, 2018-08-03:
Ok it seems to work now :+1: .