ruabmbua/hidapi-rs #26: Create basic hid report parsing module, to support usage page / number for hidraw backend

ruabmbua, 2018-08-11:
Should probably be separated into a hid report parsing crate.

jleni, 2018-08-11:
Well, I was planning to put it directly in my app because it will be something quite small.

jleni, 2018-08-22:
I have written some draft code to extract the usage_page when using Linux. It reverts back to what signal11 provides for OSX/Windows:

https://github.com/ZondaX/ledger-rs/blob/1fd8f44c155f494d50ad43d851e3084dfdbb9401/src/lib.rs#L113

Ideally, I think this additional should belong to the hidapi-rs crate and we can temporarily use it to patch the incorrect usage_page in Linux. In addition, this new code provides an example on how to move forward with a full native implementation. What do you think?

ruabmbua, 2018-08-25:
Just had a look at it, thanks! I will integrate something similar before continuing with a full implementation.
For the full implementation I am currently thinking about starting all over with a new crate. hidapi-rs on its own is already pretty useful, and I do not want to break that. Maybe even moving to a 1.0.0 release sometime soon.

zamazan4ik, 2022-03-19:
@ruabmbua any progress on this?

ruabmbua, 2022-03-19:
No, I did not continue to invest any time into this. I shifted my free time over to some projects I find more interesting.

Actually `.usage()` and `.usage_page()` methods already report the correct numbers, the hidapi C library added this feature for linux hidraw backend, and I updated the dependency.

ruabmbua, 2022-03-19:
Just verified if it works on my linux box, seems like it. If it works out for you we can close this issue.

closed 2022-04-04 by ruabmbua