ruabmbua/hidapi-rs #29: Create raw rust Windows backend (using hid.dll)

ruabmbua, 2018-08-11:
* Remove dependency to signal11/hidapi
* Match previous windows backend features

**Progress:**

- [ ] Device enumeration
- [ ] Device meta data extraction
- [ ] Blocking HID IO
- [ ] Non-blocking HID IO

jleni, 2018-08-11:
I would add appveyor CI. At least to confirm there are no build issues.

bytebuddha, 2018-08-18:
Just wanted to say, I would really love this too. 😄

AlexCouch, 2019-05-15:
I started working on a windows backend since that is what I mainly use. The current refactor was working for linux but I moved it all into cfg_if's and added the windows backend module. Also changed up the error stuff and added some comments on what I think about different backends having their own error handling implementation. The windows backend module is using setupapi from winapi since its native to windows and that is what the signal11 hidapi was using. Figured we just continue the trend. Nothing wrong with using setupapi. Also added a lot of other comments about HidApi initialization, backend initialization, error handling, and backend specific device enumeration. Basically a lot of abstraction that gets implemented per backend.

My fork's branch with all my changes can be viewed here: https://github.com/AlexCouch/hidapi-rs/tree/remove-signal11-dependency

I don't think I will be making a pull request any time soon unless its desired in the main project's repo. But it's there for tracking. I'll make a pull request when I feel as though I have the windows backend working well enough.

ruabmbua, 2019-05-16:
@AlexCouch is now working on this.

eranrund, 2022-12-01:
Hi! Wondering if there have been any updates on this? Can this crate be used on Windows?
Thanks :)

ruabmbua, 2022-12-01:
Nope no updates unfortunately. But if you want to use the crate on windows, it already works perfectly fine. The backend for windows uses the hidapi C library currently. This ticket is just about getting rid of the dependency.

eranrund, 2022-12-01:
Ah gotcha, nice. Thanks!

micolous, 2023-09-07:
Of possible interest to folks here, [I've got an implementation of the UWP API (`HidDevice`) in `webauthn-rs`](https://github.com/kanidm/webauthn-rs/blob/master/fido-hid-rs/src/windows/mod.rs), where I was able to remove its dependency on `hidapi`.

That's an MPL-2.0 licensed crate, and not designed for use outside of that particular library. Having somewhere to upstream that would be nice – but it has a lot of FIDO-specific bits in there which would need to be pulled back out, and `hidapi-rs` would also need to move to an `async`-first API (#51).

[The UWP APIs](https://learn.microsoft.com/en-us/uwp/api/windows.devices.humaninterfacedevice.hiddevice?view=winrt-22621) are pretty nice to use, they're available on Windows 10[^1] and later, and `windows-rs` bridges the COM `async` stuff across fairly nicely.

_However,_ in using it for another project, I noticed that the UWP API has a couple of limitations versus the Windows 2000-era APIs which `hidapi` uses (`HidD_*` / `HidP_*` / `hid.dll`).

* the UWP APIs can construct [AQS queries](https://learn.microsoft.com/en-us/windows/win32/search/-search-3x-advancedquerysyntax) to select HID devices (eg: `HidDevice::GetDeviceSelector`), but you can't select by VID + PID _alone_, nor matching multiple device IDs ([like macOS can](https://developer.apple.com/documentation/iokit/1438379-iohidmanagersetinputvaluematchin)).

  You can work around this by constructing the AQS by hand, but it feels a bit janky:

  ```rust
  format!("(
    System.Devices.InterfaceClassGUID:=\"{{{:?}}}\" AND
    System.Devices.InterfaceEnabled:=System.StructuredQueryType.Boolean#True AND
    System.DeviceInterface.Hid.VendorId:={vid} AND
    System.DeviceInterface.Hid.ProductId:={pid}
  )", GUID_DEVINTERFACE_HID)
  ```

* [devices with the _Consumer Usage_ are entirely unavailable](https://learn.microsoft.com/en-us/uwp/api/windows.devices.humaninterfacedevice?view=winrt-22621): `HidDevice::FromIdAsync` returns "success" and a NULL pointer, _even as an Administrator_, and even when building as a Windows Store app with the appropriate entitlements.

  You can access Consumer Usage devices with `hid.dll` just fine – even as non-Administrator. So it's an arbitrary restriction on these APIs[^2].

  This is a deal-breaker for the UWP APIs if you don't have liberty (firmware access) to change a device's descriptors. What's more frustrating is Windows doesn't have an obvious place (ie: in Device Manager or usbview) to show the HID descriptors. 😢 

Both of these are very much platform issues, but there's no obvious way to give this sort of feedback to Microsoft.

Because `hid.dll` uses fairly ordinary Windows I/O for input and output reports, it's possible to make that asynchronous (`ReadFile` / `WriteFile` with `OVERLAPPED`). `hidapi` uses those asynchronous APIs, but appears to have some pretty serious thread safety problems.

The `OVERLAPPED` bindings in `windows-rs` don't have any Rust-side `async` integration, so they're not great to use right now. But it's still possible to largely eliminate the thread-safety problems `hidapi` has today and go with a more `async`-friendly design.

[^1]: [Possibly also available on Windows 8.1](https://learn.microsoft.com/en-us/previous-versions/windows/apps/dn263140(v=win.10)#limitations-of-the-hid-api).
[^2]: There are similar restrictions for keyboards and non-Administrator access to USB HID FIDO authenticators, but those restrictions are _consistently_ applied across _all_ Windows HID APIs.

ruabmbua, 2023-09-09:
@micolous actually, we can already close this ticket. The master branch already has a native windows backend, which one can enable via cargo features. Its not yet released to crates.io, because I did not find time yet for doing some more tests to verify its working.

ruabmbua, 2023-09-09:
Regarding a more async friendly API: We can eventually do some changes to the public API surface, but it would only make sense to do that, once all our currently supported platforms get a backend not depending on hidapi, where we can implement everything with async in mind.

closed 2023-09-09 by ruabmbua