ruabmbua/hidapi-rs #33: Can't open device if vid has change?

Geobert, 2018-08-20:
Hi there,

I'm creating a HidApi at the beginning of my program and then open a device with V1/P1 as vid/pid and send a message to it to switch mode. The device is then dropped but I keep my HidApi instance.

It happens that in this mode, the vid is not the same and HipApi::open(V2, P1) fails with this error:
`Failed opening hid device`

Is that expected behavior?

ruabmbua, 2018-08-20:
You can call [HidApi::refresh_devices](https://docs.rs/hidapi/0.5.0/hidapi/struct.HidApi.html#method.refresh_devices). HidApi caches the device list from its initialization, and you have to manually refresh the list later.

This should at least allow you to discover, if the new device can actually be found.

If it is found, but you still get an error on open, then you are perhaps running into a permissions issue.

Another thing which could go wrong is, that the device is noit immediately ready. Maybe add a timeout, or watch the device list for changes.

Geobert, 2018-08-20:
Thank you for your answer!

Sorry I forgot this part: I do refresh the list (so I know the device is back online).

I've added 5s sleep to be sure but with no luck.

What kind of permission? I've tried in both way (Mode1 -> Mode2 and Mode2 -> Mode 1) and both fail on the second step :-/

Geobert, 2018-08-20:
I agree :grin:

ruabmbua, 2018-08-20:
Do you recreate the API by dropping the HidApi instance? The only thing that it really does is deinitialize the C library. 

To figure out the problem, you will have to inspect the error, or use a debugger. Anyway, what is the exact error you get?

Geobert, 2018-08-20:
https://gist.github.com/Geobert/05d3ae9e474642ba27bdd55aa009180f

here is a crude exemple

Geobert, 2018-08-20:
and the error I got is a `OpenHidDeviceError` which print into `Failed opening hid device`

ruabmbua, 2018-08-20:
Are you using HidDeviceInfo::open_device, or one of the open_XXX functions on HidApi?

ruabmbua, 2018-08-20:
I am pretty sure, that you are hitting a problem with signal11/hidapi (the underlying C library).

Geobert, 2018-08-21:
I've tried `open_device` and  `open_path` with the same issue.

ruabmbua, 2018-11-27:
I am closing this issue now.

closed 2018-11-27 by ruabmbua