ruabmbua/hidapi-rs #34: hidraw device disconnected after opening?

phl0, 2018-11-25:
I tried to use this lib and after calling the open function my hidraw device gets disconnected as the syslog says. Does this have to do with the libusb implmentation?

My code is:

```
                let api = hidapi::HidApi::new().expect(
                    "Error"
                );
                let (vid, pid) = (0x0d8c, 0x013c);
                let device = api.open(vid, pid).expect(
                    "Unable to open HID device"
                );
```

And syslog output is:

```
Nov 25 03:32:44 idefix acpid: input device has been disconnected, fd 14
Nov 25 03:32:44 idefix /usr/lib/gdm3/gdm-x-session[4192]: (II) config/udev: removing device C-Media Electronics Inc.       USB PnP Sound Device
Nov 25 03:32:44 idefix /usr/lib/gdm3/gdm-x-session[4192]: (**) Option "fd" "70"
Nov 25 03:32:44 idefix /usr/lib/gdm3/gdm-x-session[4192]: (II) event5  - C-Media Electronics Inc.       USB PnP Sound Device: device removed
Nov 25 03:32:44 idefix /usr/lib/gdm3/gdm-x-session[4192]: (II) UnloadModule: "libinput"
Nov 25 03:32:44 idefix /usr/lib/gdm3/gdm-x-session[4192]: (II) systemd-logind: releasing fd for 13:69
```

ruabmbua, 2018-11-27:
Not sure. Can you try it with the hidraw backend?

In your Cargo.toml

```toml
[dependencies.hidapi]
version = "0.5.0"
default-features = false
features = ["linux-static-hidraw"] # or linux-shared-hidraw
```

Also, does the device actually open from the perspective of your application? I mean, does api.open() return the device, and not an error?

One thing I can think of is, that libusb does not work well (or at all) with devices, which are already in use by another module / process. In your case libinput seems to use the device already. Hidraw on the other hand does work. I can open my mouse / keyboard without a problem, and both libinput and my app can read from the same device at the same time.

phl0, 2018-11-27:
Thank you for the hint. I am not a rust pro but the info seems reasonable. I am trying to port some C code to a rust application. Will test and report.

phl0, 2018-11-27:
That solved my issue. I can write to the device now. Could you please also give me a hint how to read the manufacturer and product to a string to output it in a nice way? I currently have:

```
let manufacturer = device.get_manufacturer_string().unwrap();
let product = device.get_product_string().unwrap();
println!("Found {:?} {:?}", manufacturer, product);
```

Which produces unpretty output:

```
Found Some("C-Media Electronics Inc.      ") Some("USB PnP Sound Device")
```

phl0, 2018-11-28:
Never mind. I found a way using "match".

closed 2018-11-28 by phl0