ruabmbua/hidapi-rs #38: Issues with cross-compiling hidapi for arm

phl0, 2019-04-05:
I have an issue with cross-compiling hidapi for arm architectures. Trying to compile my fork of https://github.com/rwth-afu/UniPager I always end up with the error message: 

```
--- stderr
thread 'main' panicked at 'Unable to find hidapi-libusb: CrossCompilation', src/libcore/result.rs:1009:5
note: Run with `RUST_BACKTRACE=1` for a backtrace.
```

I installed libudev-dev on my build host (Ubuntu 16.04 x86) but that did not solve the issue.
It may be an issue in combination with pkg-config. I did set some environment variables for pkg-config though:

```
declare -x PKG_ALLOW_CROSS="1"
declare -x PKG_CONFIG_DIR=""
declare -x PKG_CONFIG_LIBDIR=":/usr/lib/arm-linux-gnueabihf/"
declare -x PKG_CONFIG_SYSROOT="/"
declare -x TARGET="arm-unknown-linux-gnueabihf"
```

Any hints for further debugging this?

ruabmbua, 2019-04-05:
Can you give me a branch, where hidapi is included? I might be able to figure out the problem.

ruabmbua, 2019-04-05:
You can try building with the *linux-static-hidraw* feature. I think you might be hitting a problem, where your distro does not have multiple hidapi backends (for the C library).

On my system, and some others. the libraries are called hidapi-libusb and hidapi-hidraw, but on yours (especially the cross compile environment), the libraries might just not exist.

To link to hidapi in shared mode you need to have hidapi-libusb or hidapi-hidraw in your cross compile environment.

phl0, 2019-04-06:
The branch that I try to compile is called hidraw in my own repo. It is just a PR up to now because it doesn't compile.

Using `linux-static-hidraw`it built ok. I will check the binary and also test the travis builder.
Thanks so far for the support.

phl0, 2019-04-09:
Tested the binary successfully on RPis 1B, 2B and 3B. So using 'linux-static-hidraw' solved the problem here. Still don't understand why 'linux-shared-hidraw' doesn't work though ...

closed 2019-04-09 by phl0

lnicola, 2020-06-14:
I just got asked about this in a crate I wrote, but I'm not sure how it's supposed to work. `linux-static-hidraw` links to `libudev`, so how can I cross-compile?

As expected, I get: `/lib/x86_64-linux-gnu/libudev.so: file not recognized: file format not recognized` when trying to link a binary.