ruabmbua/hidapi-rs #45: HidError is a bit large

lnicola, 2019-08-01:
The 1.0 ship has sailed now (:tada:), but `HidError` is somewhat on the large side because of its `HidDeviceInfo` variant: 112 bytes on x64. Unfortunately, the compiler isn't too good at avoiding implicit copies (made for moves), so it's worth boxing it.

ruabmbua, 2019-08-03:
Yea thats a good idea. I will try to make a fix.

lnicola, 2019-08-03:
The problem is that the variant is public, so changing its type would technically be a breaking change. Maybe you can replace `HidDeviceInfo` with a boxed value, but..

ruabmbua, 2019-08-03:
Resolved in https://github.com/ruabmbua/hidapi-rs/commit/c2a044d7ec2036b4b63b7997523ed2041ae7025e

ruabmbua, 2019-08-03:
I just updated it to 1.0.1, I hope no one upgraded to 1.0.0 already.

closed 2019-08-03 by ruabmbua

lnicola, 2019-08-03:
I upgraded :smile:. You can yank it, I suppose, but maybe nobody is looking at the fields in the error variants. It's very easy to break semver compat this way, though.

ruabmbua, 2019-08-03:
I`ll yank 1.0.0 :P