ruabmbua/hidapi-rs #48: Feature Request: Add impl std::convert::From<hidapi::HidError> for io::Error

ConnorBP, 2019-10-19:
Currently it is not possible to use rusts ? function instead of .unwrap() due to the error that is being returned not being able to convert into the standard error. Adding this way, or some other way, to allow the use of this feature would make the syntax using the api much nicer on modern rust.

lnicola, 2019-10-19:
`hidapi::HidError` and `io::Error` are somewhat unrelated, so I'm not sure it's a good idea to provide a conversion between them, even if it was possible (I don't know how `From` and `Into` work wrt. the coherence rules).

The idiomatic solution would be to use either an `Error` enum in your crate, or something like `Box<dyn Error>`. There are a lot of error handling helpers that can help with the enum boilerplate.

ruabmbua, 2019-10-19:
As log as your function returns an error type, which defines `impl From<hidapi::Error> for YourError {}` you should be able to use the try operator.

closed 2019-10-19 by ruabmbua