ruabmbua/hidapi-rs #51: Async interface

NickeZ, 2020-02-13:
I made an [async interface to this crate](https://github.com/digitalbitbox/bitbox-bridge/tree/master/hidapi-async). However, since there is no way to register to become awakened when there is data to read I have to create a thread and do blocking reads.

Reading through libhidapi (the C library) I realised that the libusb interface actually is asynchronous and that libhidapi conveniently (unfortunately) wraps away that using an internal thread. So the result feels incredible clunky with two threads and multiple mutexes / synchronization primitives.

I think the best approach actually is to skip hidapi and rewrite using the platform specific bindings from scratch. But before I start such a big effort I wanted to check in here if anyone else already is working on it.

Thanks

ruabmbua, 2020-02-14:
Hi @NickeZ .

I was already working on a raw rust backend for this crate, which would skip hidapi (the C library), and use the os interfaces directly.

At the time I did this, I also wanted to add async support, but the rust async story was not really stable. I decided to wait, until async ecosystem / language integration catches up. I have not looked into async / await since then, but from what I heard, it might be ready

Regarding libusb, I do not plan on using it at all. All somewhat modern Linux systems should have the hidraw kernel interface.

I might be able to spend some time on this starting next week, but I have some other hobby projects going on.

https://github.com/ruabmbua/hidapi-rs/projects/1

ruabmbua, 2020-02-14:
If you have any pointers, if I should try it with the tokio ecosystem, or async-std, please tell me.
Maybe it`s possible to be runtime independent by just using mio, std::future and std::task?

NickeZ, 2020-02-16:
I haven't looked close enough to know exactly what crates are needed, but I think you shouldn't need `tokio`. I think technically it should be enough with `std::future`, `std::task` and `AsyncRead/Write` from `futures`. However it would be great if we could reuse some platform specific code from `mio` to avoid reimplementing it. But it sounds to me like that depends on how usb works on the different platforms.

One caveat is that we may have to implement `AsyncRead/Write` from `tokio` as well to support that ecosystem. Currently it is in a bit of flux how the final std traits will look.

lnicola, 2020-02-16:
Tokio will migrate to the version of the async traits that gets standardized, but until then it would be nice to support both, if possible.

NickeZ, 2020-02-16:
So the plan would be to wrap these interfaces?

* Windows (using hid.dll)
* Linux (using the Kernel's hidraw driver)
* Mac (using IOHidManager)

brokenthorn, 2020-02-17:
> So the plan would be to wrap these interfaces?
> 
> * Windows (using hid.dll)
> * Linux (using the Kernel's hidraw driver)
> * Mac (using IOHidManager)

For Windows, there's [winapi-rs](https://github.com/retep998/winapi-rs) which provides raw FFI bindings to all of Windows API.

Yamakaky, 2020-06-12:
In the mean time, I think we could use https://docs.rs/smol as a hidapi async wrapper. The only thing we would need would be to make HidDevice AsRawFd (on linux at least). For testing, we can get the fd by casting the device handle as it is the first member (https://github.com/libusb/hidapi/blob/master/linux/hid.c#L65). A more robust solution would be to patch hidapi with a method returning the device specific fd.

brokenthorn, 2020-08-03:
What about [rust/winrt](https://lib.rs/crates/winrt)? Does it supersede this crate?

lnicola, 2020-08-03:
@brokenthorn WinRT might have some USB functions, but it doesn't deprecate this crate which is portable across Windows, Linux and MacOS.

ruabmbua, 2020-08-06:
Don`t forget BSD :P

timfish, 2021-10-09:
Just dropping in here to mention that it looks like the `winapi` crate has effectively been superseded by the official Microsoft `windows` crate!

darkdragon-001, 2022-04-26:
Any news on async hidraw on Linux?

NickeZ, 2022-04-26:
Nope sorry, I don't work with usb hardware right now so I'm not looking into it anymore.

brokenthorn, 2022-05-08:
> What about [rust/winrt](https://lib.rs/crates/winrt)? Does it supersede this crate?

That crate has now been deprecated in favor of [the windows crate](https://lib.rs/crates/windows) which is wider reaching and might now expose more of the Windows APIs.

mcuee, 2023-05-14:
> Don`t forget BSD :P

It is interesting to see such efforts to bypass hidapi. :-)

In the case of FreeBSD, take note there is the new hidraw (similar to hidraw under Linux) for newer version of FreeBSD.

Reference: the following hidraw feature has been merged.
https://reviews.freebsd.org/D27992

zardini123, 2024-02-02:
I came across this thread as I am also interested in async HID in Rust.  Doing a search, I found someone created a async HID crate using platform-native bindings called [async-hid](https://crates.io/crates/async-hid).  The library seems rather unknown (73 downloads all time), and entire commit history is 3 months (July 8th, 2023 to Oct 12, 2023).  With that, I have no idea the state of support for the project.

Has anyone here given that project a shot?

carlosmn, 2026-01-01:
I polished off some async work I've had in around in a few different variants and opened #178 for the easier backend of linux-native. I expect/hope that other native implementations would also make it relatively easy and maybe you don't even need to choose a runtime.

On the libusb side, I looked into reimplementing hidapi's logic but it does quite a bit and there's some stuff around xbox controllers and languages that I wouldn't want to have to maintain twice.

As much as I dislike what might be the only option, implementing async for the hidapi backend should give you support across all the environments. Unfortunately what we'd have to do is a bit gross.

hidapi is already doing async reads behind the scenes but you can only get access to it via `hid_read` or `hid_read_timeout`. I had considered trying to extract the libusb device handle from under it and call things via rusb ourselves, but it starts its read thread as soon as you open the device, so there is no space for us to get extract the handle before it starts its loop of reading from the device.

Perhaps you could get in its `hid_device` and flip `shutdown_thread` to `1` so it doesn't retry and then you can use `hid_read` until the queue is empty and then switch to doing the transfers ourselves. I don't know if we actually can do this reliably but maybe it's worth a try. It does however mean that we have to do all reads ourselves, even the sync ones. That part probably wouldn't be the worst.

Alternatively, the safe (in most senses) thing to do is to spawn a thread that essentially does the same thing hidapi does with libusb and in `poll_read()` we'd have to spawn a thread to wait for `hid_read_timeout` to return, then that can store the result somewhere and signal for wakeup. A big issue here is that we cannot cancel this read unless we close the device. We would have to take these reads into consideration even in sync reads in case the caller gave up on the async read (e.g. via timeout) and they're now trying a sync one (probably we just need to see if the slot is full and return that instead of going to hidapi).

Writing is where this gets a bit more unfortunate. We do not have a way to do asynchronous writes into hidapi. The way to do this is (like hidapi does for reads) to issue a transfer and provide a callback. For this it does look like we'd either have to open the device ourselves again or extract the pointer from `hid_device`. Opening a device multiple times sounds like it's not something you can always rely on but I don't know what the deal with that is on macOS.

But even if we do get the handle, AFAICT rusb doesn't have support for async sending of transfers so we'd have to either call libusb directly or first implement it as part of rusb.