ruabmbua/hidapi-rs #55: hidraw build error

fpagliughi, 2020-04-08:
I'm trying to build on Linux (Mint 19) using the _hidraw_ backend, but I'm getting a build error.

```
[dependencies.hidapi]
version = "1.2"
default-features = false
features = ["linux-static-hidraw"]
```

```
error: failed to run custom build command for `hidapi v1.2.1`

Caused by:
  process didn't exit successfully: `/home/fmp/customers/tulip/repos/hidd/environments/kado/hw/hidd/target/debug/build/hidapi-432a5c23f3442b71/build-script-build` (exit code: 101)
--- stderr
thread 'main' panicked at 'Exactly one linux hidapi backend must be selected.', /home/fmp/.cargo/registry/src/github.com-1ecc6299db9ec823/hidapi-1.2.1/build.rs:94:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

warning: build failed, waiting for other jobs to finish...
```

Am I doing something wrong?

ruabmbua, 2020-04-09:
Hm. It is working in my case.

ruabmbua, 2020-04-09:
Are you maybe including hidapi as a transitive dependency somewhere else? One of your dependencies might request another hidapi backend.

fpagliughi, 2020-04-13:
Yes! That was it. Thank you.

closed 2020-04-13 by fpagliughi