ruabmbua/hidapi-rs #63: hidapi 1.2.4 : build errors (linker errors) on macOS 10.15.7

japaric, 2020-11-26:
Hello, `probe-run` developer here.

Our users have reported us build errors when running `cargo install probe-run --version 0.1.6`. See below:

``` console
  = note: Undefined symbols for architecture x86_64:
            "_NSAppKitVersionNumber", referenced from:
                _hid_init in libhidapi-6c82a0d52f552184.rlib(hid.o)
                _hid_enumerate in libhidapi-6c82a0d52f552184.rlib(hid.o)
                _hid_open_path in libhidapi-6c82a0d52f552184.rlib(hid.o)
          ld: symbol(s) not found for architecture x86_64
          clang: error: linker command failed with exit code 1 (use -v to see invocation)

error: aborting due to previous error

error: failed to compile `probe-run v0.1.6 (https://github.com/knurling-rs/probe-run?branch=main#013c0bd3)`, intermediate artifacts can be found at `/var/folders/5b/jlxfj70x1fb_n2l6qh21xxfh0000gn/T/cargo-installcEziMp`

Caused by:
  could not compile `probe-run`.

To learn more, run the command again with --verbose.
```

original report: knurling-rs/probe-run#113

we have narrowed down the issue to the recent hidapi update (1.2.3 -> 1.2.4) and are currently working around the issue by pinning our hidapi dependency to v1.2.3 knurling-rs/probe-run#114

we have, however, been unsuccessful in producing a minimal repro case. The following steps:

``` rust
$ cargo new --bin hello
$ cd hello
```

``` rust
// src/main.rs
use hidapi as _;
fn main() {}
```

``` toml
# Cargo.toml
[dependencies]
hidapi = "1.2.4"
```

``` console
$ cargo build
```

produce a crate that *sometimes* fails to link but usually builds fine. The linker errors have been observed with both Rust 1.46 and 1.48

P.S. I'm not a macOS user myself

ruabmbua, 2020-11-26:
Well, this took only 1 day to bite me back. I published the new version yesterday without finishing CI tests, because travis CI was not working (all CI jobs were stalled for a long time, and I just cancelled it). Maybe because of the AWS outage?

Anyway, it should work on windows, and Linux, because I tested on both on my own machines, but I do not have a mac.

I will try to take a look on the macos issue later today, but no promises, I don`t own a mac so this may become somewhat difficult.
I will yank the release in the meantime.

andresv, 2020-11-27:
So it is fixed in `1.2.5` https://github.com/ruabmbua/hidapi-rs/commit/76e0f904e5151179d0c927b9d06a5600702d388f.

ruabmbua, 2020-11-27:
Yes, forgot to close the issue.

closed 2020-11-27 by ruabmbua