ruabmbua/hidapi-rs #65: Build fails on Linux Mint 20

pantonov, 2020-12-06:
Build script fails to find libusb, but libusb-dev and libusb-1.0 are installed from default repository,  both `/usr/bin/libusb-config` and `/usr/lib/x86_64-linux-gnu/pkgconfig/libusb.pc` are present. This affects installation of popular crates such as `cargo-flash`.

The current workaround is to manually download/compile/install libusb from [github.com/libusb](github.com/libusb).
```
thread 'main' panicked at 'Unable to find libusb-1.0: "`\"pkg-config\" \"--libs\" \"--cflags\" \"libusb-1.0\"` did not exit successfully: exit code: 1\n--- stderr\nPackage libusb-1.0 was not found in the pkg-config search path.\nPerhaps you should add the directory containing `libusb-1.0.pc\'\nto the PKG_CONFIG_PATH environment variable\nNo package \'libusb-1.0\' found\n"', /home/apg/.cargo/registry/src/github.com-1ecc6299db9ec823/hidapi-1.2.5/build.rs:65:60
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

```
# apt list libusb-dev libusb-1.0-0
Listing... Done
libusb-1.0-0/focal,now 2:1.0.23-2build1 amd64 [installed]
libusb-1.0-0/focal,now 2:1.0.23-2build1 i386 [installed,automatic]
libusb-dev/focal,now 2:0.1.12-32 amd64 [installed]
libusb-dev/focal 2:0.1.12-32 i386
```

ruabmbua, 2020-12-09:
You have to install libusb-1.0.0-dev as well.

pantonov, 2020-12-10:
The proper package name is apparently `libusb-1.0-0-dev`, which is somewhat confusing and I think should be mentioned in installation instructions. README's for cargo-flash etc. tell just to  `apt install libusb-dev`, which is apparently not enough.

```
# apt install libusb-1.0.0-dev
...
E: Unable to locate package libusb-1.0.0-dev

# apt-get install libusb-1.0.0-dev
...
Note, selecting 'libusb-1.0-0-dev' for regex 'libusb-1.0.0-dev'
The following additional packages will be installed:
  libusb-1.0-doc
The following NEW packages will be installed:
  libusb-1.0-0-dev libusb-1.0-doc
```

ruabmbua, 2020-12-10:
Maybe create an issue in cargo-flash for updating the README with libusb-1.0.0-dev?

closed 2020-12-15 by ruabmbua

vyorkin, 2025-04-12:
Thanks for mentioning libusb-1.0.0-dev at least here :) Fixed the build for me