ruabmbua/hidapi-rs #66: VID/PID is incorrect?

vektrixdev, 2020-12-15:
I'm trying to write a driver for the oculus quest (link). 

Looking at the device list (`api.devices()`), All the VIDs seem incorrect (ie `1241` (belkin), yet I don't have any belkin devices connected). 

I see several devices with PID `0137`, and vid `9494`, which is the quest's PID, but I can't confirm as the vid is incorrect.

ruabmbua, 2020-12-15:
Which platform are you targeting?

Need a little bit more info.

vektrixdev, 2020-12-15:
I'm targeting `x86_64-unknown-linux-gnu`.
I should have noted that rusb sees the correct info.

ruabmbua, 2020-12-15:
Can you test the different backends?

Maybe try the linux-static-hidraw backend instead of the libusb one.

vektrixdev, 2020-12-15:
Tried it, same problem.

ruabmbua, 2020-12-15:
That is certainly weird. 

Can you clone the repository and run `cargo run --example lshid --no-default-features --features linux-static-hidraw` as a test?

vektrixdev, 2020-12-15:
Sorry, I was mixing up hex and decimal. My bad.

closed 2020-12-15 by vektrixdev