ruabmbua/hidapi-rs #68: HidApiError { message: "hid_error is not implemented yet" }'

legendofmiracles, 2021-03-04:
So i want to communicate with my keyboard, but whenever i want to send data, it crashes. 
```
❯ cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.00s
     Running `target/debug/keeb`
33
33
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: HidApiError { message: "hid_error is not implemented yet" }', src/main.rs:12:33
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

```
with the following code: 
```rust
use hidapi::HidApi;
fn main() {
    let api = HidApi::new().unwrap();
    let keeb = api.open(0x3297, 0x4974);
    if keeb.is_ok() {
        let keeb = keeb.unwrap();
        // keycodes 
        let codes = vec![28, 33, 38, 43, 47, 0, 5, 10, 15];
        // payload has to be 32 bytes + 1 control byte
        let mut buffer = vec![0; 33];
        // right now taking the first number, will be replaced later  
        buffer[1] = codes[1 - 1];
        println!("{:#?}", std::mem::size_of::<[u8; 33]>());
        println!("{:#?}", std::mem::size_of_val(&buffer[..]));
        keeb.write(&buffer[..]).unwrap();
        println!("It worked!");
    }
}

```
What could be causing this?

ruabmbua, 2021-03-04:
Can you tell me your platform, and version of the library (Cargo.lock file should be enough).

ruabmbua, 2021-03-04:
Also backtrace would be helpful, is line 12 the keeb.write(), or something else?

ruabmbua, 2021-03-04:
It looks like you are either using mac, or some platform with libusb. In that case, acquiring a meaningful error message is not implemented yet.

If you use linux, I recommend switching to the hidraw backend instead of usb, if you are running on a kernel, which is not totally outdated.

To use the hidraw backend on linux, deactivate default features, and activate one "linux-static-hidraw", or "linux-shared-hidraw" (see Cargo.toml).

legendofmiracles, 2021-03-05:
I'm on archlinux, on the 5.11 kernel. So not a old platform.

But I did write a python script, which should do the identical thing and it always required root to run, so maybe that's the issue.When i run this binary as root it also doesn't help.

```
    Finished dev [unoptimized + debuginfo] target(s) in 0.00s
     Running `target/debug/keeb`
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: HidApiError { message: "hid_error is not implemented yet" }', src/main.rs:12:33
stack backtrace:
   0:     0x558a5b074160 - std::backtrace_rs::backtrace::libunwind::trace::h25e12e0d899beba0
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/../../backtrace/src/backtrace/libunwind.rs:90:5
   1:     0x558a5b074160 - std::backtrace_rs::backtrace::trace_unsynchronized::h70e61195d6ae3df6
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
   2:     0x558a5b074160 - std::sys_common::backtrace::_print_fmt::hba93ab80d779695a
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/sys_common/backtrace.rs:67:5
   3:     0x558a5b074160 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::hf092b5883b4b2e50
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/sys_common/backtrace.rs:46:22
   4:     0x558a5b08ad9c - core::fmt::write::hf68bc350a8f2f0dc
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/core/src/fmt/mod.rs:1078:17
   5:     0x558a5b072252 - std::io::Write::write_fmt::hf66811b1bc767436
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/io/mod.rs:1517:15
   6:     0x558a5b075e95 - std::sys_common::backtrace::_print::hd425a11bfe1f20f8
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/sys_common/backtrace.rs:49:5
   7:     0x558a5b075e95 - std::sys_common::backtrace::print::h6d678795c1e61e13
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/sys_common/backtrace.rs:36:9
   8:     0x558a5b075e95 - std::panicking::default_hook::{{closure}}::h78a02a4a0dee5e7e
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/panicking.rs:208:50
   9:     0x558a5b0759ea - std::panicking::default_hook::h56eb7eda02f355a7
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/panicking.rs:225:9
  10:     0x558a5b076631 - std::panicking::rust_panic_with_hook::hb27ea14285131c61
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/panicking.rs:591:17
  11:     0x558a5b076177 - std::panicking::begin_panic_handler::{{closure}}::hc552fcee62aad17f
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/panicking.rs:497:13
  12:     0x558a5b07461c - std::sys_common::backtrace::__rust_end_short_backtrace::hb9f0aa9a78e885a0
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/sys_common/backtrace.rs:141:18
  13:     0x558a5b0760d9 - rust_begin_unwind
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/panicking.rs:493:5
  14:     0x558a5b08a171 - core::panicking::panic_fmt::h12ac4570ea43d06f
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/core/src/panicking.rs:92:14
  15:     0x558a5b089f93 - core::option::expect_none_failed::h096fa60f757b7204
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/core/src/option.rs:1268:5
  16:     0x558a5b045b63 - core::result::Result<T,E>::unwrap::h6feb3832e8ee6f26
                               at /home/legend/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/result.rs:973:23
  17:     0x558a5b04634c - keeb::main::h7226eb8d77f2f3d2
                               at /home/legend/keeb/src/main.rs:12:9
  18:     0x558a5b046c9b - core::ops::function::FnOnce::call_once::hba4bef55b32f1477
                               at /home/legend/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ops/function.rs:227:5
  19:     0x558a5b046a9e - std::sys_common::backtrace::__rust_begin_short_backtrace::h7b7a82a0dabbc097
                               at /home/legend/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/sys_common/backtrace.rs:125:18
  20:     0x558a5b045f41 - std::rt::lang_start::{{closure}}::h0737237a2de769f3
                               at /home/legend/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/rt.rs:66:18
  21:     0x558a5b076a57 - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once::h78040f802d89ccdc
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/core/src/ops/function.rs:259:13
  22:     0x558a5b076a57 - std::panicking::try::do_call::h6853cad536dd09a1
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/panicking.rs:379:40
  23:     0x558a5b076a57 - std::panicking::try::h827495f03a9fbb9a
[... cut ...]
                               at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/rt.rs:51:25
  26:     0x558a5b045f17 - std::rt::lang_start::hd6a86e7eb75980c8
                               at /home/legend/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/rt.rs:65:5
  27:     0x558a5b0464ea - main
  28:     0x7f077e712b25 - __libc_start_main
  29:     0x558a5b0453ae - _start
  30:                0x0 - <unknown>
````

Line 12 is keeb.write, yes.

legendofmiracles, 2021-03-05:
EDIT: I switched to the other backend. And now it has the similar behavior like the python lib. 
[image]
*at least a bit more*

ruabmbua, 2021-03-12:
Any update on this, or are you just experiencing problems with your specific device?

closed 2021-03-16 by ruabmbua