ruabmbua/hidapi-rs #69: Static CRT option for Windows

pravic, 2021-04-27:
It would be nice if this crate allowed to link C runtime library statically, as a cargo feature option (say, `windows-static-crt`).

https://github.com/ruabmbua/hidapi-rs/blob/06354516eb65a65319748c28ea622b3eb324005a/build.rs#L115-L121

 needs to be modified to include a call to `static_crt(true)` 

See https://docs.rs/cc/1.0.67/cc/struct.Build.html#method.static_crt

ruabmbua, 2021-04-27:
Is it a good idea, to link the crt statically? According to my windows knowledge (which might be horribly / out of date), the syscall numbers etc... are very unstable in the windows kernel. The crt is probably using some, so it might cause incompatibilities?

Of course, I would gladly review a PR, if you have the time to create one.

pravic, 2021-04-27:
It has nothing to do with syscalls - CRT on Windows is about [C runtime](https://docs.microsoft.com/en-us/cpp/c-runtime-library/crt-library-features?view=msvc-160). Syscalls are managed by [NTDLL](https://en.wikipedia.org/wiki/Microsoft_Windows_library_files#NTDLL.DLL).

pravic, 2021-04-27:
As for the PR - okay, I'll make one.

PR #70 (2021-04-28): Static CRT support
Tested, works fine.

It builds static CRT only when required: https://doc.rust-lang.org/reference/linkage.html#static-and-dynamic-c-runtimes.

Fixes #69.
closed 2021-04-28 by ruabmbua