ruabmbua/hidapi-rs #76: Build error at hidapi v1.3.2

marior765, 2022-01-15:
Error when building, Mac m1 Big Sur 11.4, need some help there please

`error: failed to run custom build command for `hidapi v1.3.2`

Caused by:
  process didn't exit successfully: `/Users/*/projects/*/target/debug/build/hidapi-a6c2c51c7495521e/build-script-build` (exit status: 1)
  --- stdout
  TARGET = Some("aarch64-apple-darwin")
  OPT_LEVEL = Some("0")
  HOST = Some("aarch64-apple-darwin")
  CC_aarch64-apple-darwin = None
  CC_aarch64_apple_darwin = None
  HOST_CC = None
  CC = None
  CFLAGS_aarch64-apple-darwin = None
  CFLAGS_aarch64_apple_darwin = None
  HOST_CFLAGS = None
  CFLAGS = None
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some("true")
  CARGO_CFG_TARGET_FEATURE = None
  running: "cc" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-fno-omit-frame-pointer" "-arch" "arm64" "-I" "etc/hidapi/hidapi" "-Wall" "-Wextra" "-o" "/Users/*/projects/*/target/debug/build/hidapi-e39b64c02a020d45/out/etc/hidapi/mac/hid.o" "-c" "etc/hidapi/mac/hid.c"
  cargo:warning=In file included from etc/hidapi/mac/hid.c:25:
  cargo:warning=In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDManager.h:27:
  cargo:warning=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOTypes.h:81:49: error: expected ';' after top level declarator
  cargo:warning=typedef mach_vm_address_t       IOVirtualAddress __kernel_ptr_semantics;
  cargo:warning=                                                ^
  cargo:warning=In file included from etc/hidapi/mac/hid.c:27:
  cargo:warning=In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOKitLib.h:52:
  cargo:warning=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/OSMessageNotification.h:120:53: error: expected ';' after top level declarator
  cargo:warning=typedef natural_t OSAsyncReference[kOSAsyncRefCount] __kernel_ptr_semantics;
  cargo:warning=                                                    ^
  cargo:warning=In file included from etc/hidapi/mac/hid.c:32:
  cargo:warning=/usr/local/include/pthread.h:328:6: warning: macro expansion producing 'defined' has undefined behavior [-Wexpansion-to-defined]
  cargo:warning=#if !_PTHREAD_SWIFT_IMPORTER_NULLABILITY_COMPAT
  cargo:warning=     ^
  cargo:warning=/usr/local/include/pthread.h:197:2: note: expanded from macro '_PTHREAD_SWIFT_IMPORTER_NULLABILITY_COMPAT'
  cargo:warning=        defined(SWIFT_CLASS_EXTRA) && (!defined(SWIFT_SDK_OVERLAY_PTHREAD_EPOCH) || (SWIFT_SDK_OVERLAY_PTHREAD_EPOCH < 1))
  cargo:warning=        ^
  cargo:warning=/usr/local/include/pthread.h:328:6: warning: macro expansion producing 'defined' has undefined behavior [-Wexpansion-to-defined]
  cargo:warning=/usr/local/include/pthread.h:197:34: note: expanded from macro '_PTHREAD_SWIFT_IMPORTER_NULLABILITY_COMPAT'
  cargo:warning=        defined(SWIFT_CLASS_EXTRA) && (!defined(SWIFT_SDK_OVERLAY_PTHREAD_EPOCH) || (SWIFT_SDK_OVERLAY_PTHREAD_EPOCH < 1))
  cargo:warning=                                        ^
  cargo:warning=/usr/local/include/pthread.h:537:6: warning: macro expansion producing 'defined' has undefined behavior [-Wexpansion-to-defined]
  cargo:warning=#if !_PTHREAD_SWIFT_IMPORTER_NULLABILITY_COMPAT
  cargo:warning=     ^
  cargo:warning=/usr/local/include/pthread.h:197:2: note: expanded from macro '_PTHREAD_SWIFT_IMPORTER_NULLABILITY_COMPAT'
  cargo:warning=        defined(SWIFT_CLASS_EXTRA) && (!defined(SWIFT_SDK_OVERLAY_PTHREAD_EPOCH) || (SWIFT_SDK_OVERLAY_PTHREAD_EPOCH < 1))
  cargo:warning=        ^
  cargo:warning=/usr/local/include/pthread.h:537:6: warning: macro expansion producing 'defined' has undefined behavior [-Wexpansion-to-defined]
  cargo:warning=/usr/local/include/pthread.h:197:34: note: expanded from macro '_PTHREAD_SWIFT_IMPORTER_NULLABILITY_COMPAT'
  cargo:warning=        defined(SWIFT_CLASS_EXTRA) && (!defined(SWIFT_SDK_OVERLAY_PTHREAD_EPOCH) || (SWIFT_SDK_OVERLAY_PTHREAD_EPOCH < 1))
  cargo:warning=                                        ^
  cargo:warning=4 warnings and 2 errors generated.
  exit status: 1

  --- stderr

  error occurred: Command "cc" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-fno-omit-frame-pointer" "-arch" "arm64" "-I" "etc/hidapi/hidapi" "-Wall" "-Wextra" "-o" "/Users/*/*/*/target/debug/build/hidapi-e39b64c02a020d45/out/etc/hidapi/mac/hid.o" "-c" "etc/hidapi/mac/hid.c" with args "cc" did not execute successfully (status code exit status: 1).

`

marior765, 2022-01-17:
If manually downgrade the Xcode from 13.2.1 to 12.5.1 - problem is gone

ruabmbua, 2022-01-17:
Thanks for the report! Unfortunately I have no M1 mac available for troubleshooting. Maybe its the same issue on new xcode + intel mac?

And can you try out to compile the hid.c file with the xcode C compiler?

marior765, 2022-01-19:
@ruabmbua not sure if my doing everything right but I got this
`Apple clang version 12.0.5 (clang-1205.0.22.11)
Target: arm64-apple-darwin20.5.0
Thread model: posix`
//
`Undefined symbols for architecture arm64:
  "_CFArrayGetCount", referenced from:
      _create_device_info in hid-a6976b.o
  "_CFArrayGetTypeID", referenced from:
      _get_array_property in hid-a6976b.o
  "_CFArrayGetValueAtIndex", referenced from:
      _create_device_info in hid-a6976b.o
  "_CFDictionaryCreateMutable", referenced from:
      _hid_enumerate in hid-a6976b.o
  "_CFDictionaryGetTypeID", referenced from:
      _create_device_info in hid-a6976b.o
  "_CFDictionaryGetValueIfPresent", referenced from:
      _create_device_info in hid-a6976b.o
  "_CFDictionarySetValue", referenced from:
      _hid_enumerate in hid-a6976b.o
  "_CFGetTypeID", referenced from:
      _create_device_info in hid-a6976b.o
      _get_int_property in hid-a6976b.o
      _get_array_property in hid-a6976b.o
  "_CFNumberCreate", referenced from:
      _hid_enumerate in hid-a6976b.o
  "_CFNumberGetTypeID", referenced from:
      _create_device_info in hid-a6976b.o
      _get_int_property in hid-a6976b.o
  "_CFNumberGetValue", referenced from:
      _create_device_info in hid-a6976b.o
      _get_int_property in hid-a6976b.o
  "_CFRelease", referenced from:
      _hid_exit in hid-a6976b.o
      _hid_enumerate in hid-a6976b.o
      _hid_open_path in hid-a6976b.o
      _free_hid_device in hid-a6976b.o
      _hid_close in hid-a6976b.o
  "_CFRunLoopAddSource", referenced from:
      _read_thread in hid-a6976b.o
  "_CFRunLoopGetCurrent", referenced from:
      _init_hid_manager in hid-a6976b.o
      _read_thread in hid-a6976b.o
  "_CFRunLoopGetMain", referenced from:
      _hid_close in hid-a6976b.o
  "_CFRunLoopRunInMode", referenced from:
      _process_pending_events in hid-a6976b.o
      _read_thread in hid-a6976b.o
  "_CFRunLoopSourceCreate", referenced from:
      _read_thread in hid-a6976b.o
  "_CFRunLoopSourceSignal", referenced from:
      _hid_close in hid-a6976b.o
  "_CFRunLoopStop", referenced from:
      _hid_device_removal_callback in hid-a6976b.o
      _perform_signal_callback in hid-a6976b.o
  "_CFRunLoopWakeUp", referenced from:
      _hid_close in hid-a6976b.o
  "_CFSetGetCount", referenced from:
      _hid_enumerate in hid-a6976b.o
  "_CFSetGetValues", referenced from:
      _hid_enumerate in hid-a6976b.o
  "_CFStringCreateWithCString", referenced from:
      _hid_open_path in hid-a6976b.o
  "_CFStringGetBytes", referenced from:
      _get_string_property in hid-a6976b.o
  "_CFStringGetLength", referenced from:
      _get_string_property in hid-a6976b.o
  "_IOHIDDeviceClose", referenced from:
      _hid_close in hid-a6976b.o
  "_IOHIDDeviceCreate", referenced from:
      _hid_open_path in hid-a6976b.o
  "_IOHIDDeviceGetProperty", referenced from:
      _get_int_property in hid-a6976b.o
      _get_array_property in hid-a6976b.o
      _get_string_property in hid-a6976b.o
  "_IOHIDDeviceGetReport", referenced from:
      _get_report in hid-a6976b.o
  "_IOHIDDeviceGetService", referenced from:
      _create_device_info_with_usage in hid-a6976b.o
  "_IOHIDDeviceOpen", referenced from:
      _hid_open_path in hid-a6976b.o
  "_IOHIDDeviceRegisterInputReportCallback", referenced from:
      _hid_open_path in hid-a6976b.o
      _hid_close in hid-a6976b.o
  "_IOHIDDeviceRegisterRemovalCallback", referenced from:
      _hid_open_path in hid-a6976b.o
      _hid_close in hid-a6976b.o
  "_IOHIDDeviceScheduleWithRunLoop", referenced from:
      _read_thread in hid-a6976b.o
      _hid_close in hid-a6976b.o
  "_IOHIDDeviceSetReport", referenced from:
      _set_report in hid-a6976b.o
  "_IOHIDDeviceUnscheduleFromRunLoop", referenced from:
      _hid_close in hid-a6976b.o
  "_IOHIDManagerClose", referenced from:
      _hid_exit in hid-a6976b.o
  "_IOHIDManagerCopyDevices", referenced from:
      _hid_enumerate in hid-a6976b.o
  "_IOHIDManagerCreate", referenced from:
      _init_hid_manager in hid-a6976b.o
  "_IOHIDManagerScheduleWithRunLoop", referenced from:
      _init_hid_manager in hid-a6976b.o
  "_IOHIDManagerSetDeviceMatching", referenced from:
      _init_hid_manager in hid-a6976b.o
      _hid_enumerate in hid-a6976b.o
  "_IOObjectRelease", referenced from:
      _hid_open_path in hid-a6976b.o
  "_IORegistryEntryFromPath", referenced from:
      _hid_open_service_registry_from_path in hid-a6976b.o
  "_IORegistryEntryGetRegistryEntryID", referenced from:
      _create_device_info_with_usage in hid-a6976b.o
  "_IORegistryEntryIDMatching", referenced from:
      _hid_open_service_registry_from_path in hid-a6976b.o
  "_IOServiceGetMatchingService", referenced from:
      _hid_open_service_registry_from_path in hid-a6976b.o
  "_NSAppKitVersionNumber", referenced from:
      _hid_init in hid-a6976b.o
  "___CFConstantStringClassReference", referenced from:
      CFString in hid-a6976b.o
      CFString in hid-a6976b.o
      CFString in hid-a6976b.o
      CFString in hid-a6976b.o
      CFString in hid-a6976b.o
      CFString in hid-a6976b.o
      CFString in hid-a6976b.o
      ...
  "_kCFAllocatorDefault", referenced from:
      _init_hid_manager in hid-a6976b.o
      _hid_enumerate in hid-a6976b.o
      _hid_open_path in hid-a6976b.o
      _read_thread in hid-a6976b.o
  "_kCFRunLoopDefaultMode", referenced from:
      _init_hid_manager in hid-a6976b.o
      _process_pending_events in hid-a6976b.o
      _hid_close in hid-a6976b.o
  "_kCFTypeDictionaryKeyCallBacks", referenced from:
      _hid_enumerate in hid-a6976b.o
  "_kCFTypeDictionaryValueCallBacks", referenced from:
      _hid_enumerate in hid-a6976b.o
  "_kIOMasterPortDefault", referenced from:
      _hid_open_service_registry_from_path in hid-a6976b.o
  "_main", referenced from:
     implicit entry/start for main executable
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)`

ruabmbua, 2022-01-20:
Its interesting, that it is only a linker error this time. Probably because you need to specify the proper libs (frameworks in macos), for it to work. Unfortunately I won`t be able to help much, without access to a mac.

ruabmbua, 2022-02-10:
@marior765 could you try again if the compile of the crate works with the newest version?
There were some updates on the underling hidapi C library, and I updated to the latest version.

ruabmbua, 2022-05-18:
Please create a new ticket if this issue is still happening.
Closed because of missing response.

closed 2022-05-18 by ruabmbua