ruabmbua/hidapi-rs #81: Multiple open on same device leads to device lock

becominginsane, 2022-03-31:
A [report](https://github.com/Zondax/ledger-rs/issues/30) was made to a crate using `hidapi` where the user was looking to open a device multiple times but incurring in weird behavior.

I've added some new APIs to the crate to allow this, and during testing for that feature my device froze after closing the device twice.
What I mean is opening the same device multiple times, and then dropping the handles together, so to have N open instances at once...

I think this leads to a "double free" in the underlying library as the device becomes unresponsive until power cycled (could also be an issue with the device itself!)

ruabmbua, 2022-04-03:
To which crate did you have to add new API`s? To hidapi-rs? If so, please post it as a draft PR, then I can take a look.

becominginsane, 2022-04-03:
In the original message I had linked the project's issue and I've also metioned this issue there so you can also see it in github.

The project is [Zondax/ledger-rs](https://github.com/Zondax/ledger-rs), specifically the `ledger-transport-hid` crate. I mentioned I added new APIs to allow opening different devices but that was to `ledger-transport-hid` crate.

There's a test that I have marked with `#[ignore]` until this device lock issue is resolved: https://github.com/Zondax/ledger-rs/blob/cc4a966a47db9cc885a5032fb9ed44dc288e97fa/ledger-transport-hid/src/lib.rs#L297-L332

0xcacti, 2023-10-06:
Any update here?  Would be really nice to be able to implement the clone trait in https://github.com/Zondax/ledger-rs 
If it's straightforward, I am happy to try to extend the hidapi-rs, but not sure where to start.

liugangnhm, 2024-11-30:
TcpStream::try_clone like api would be very nice