ruabmbua/hidapi-rs #84: Can not open BlueTooth Hid device on Mac

dongjian, 2022-06-24:
hidapi-rs version:
hidapi = "1.4.1"
system:
macOS 11.6.4
Device:
BlueTooth 3.0/5.0 Keyboard 
test code :
hidapi.device_list().for_each(|v| {
            if v.product_id() == 0x4013 {
                println!(
                    "---vid  {:x?}  pid  {:x?}  usage  {:x?}   usagepage{:x?}",
                    v.product_id(),
                    v.vendor_id(),
                    v.usage(),
                    v.usage_page()
                );
                let res = v.open_device(&hidapi);
                match res {
                    Ok(_) => {
                        println!("OOOPPPPEEEENNN!!!!!!")
                    }
                    Err(_) => (),
                };
            }
        });
===
log:
---vid  4013  pid  3151  usage  6   usagepage1
---vid  4013  pid  3151  usage  1   usagepagec
---vid  4013  pid  3151  usage  80   usagepage1
---vid  4013  pid  3151  usage  2   usagepage1
---vid  4013  pid  3151  usage  1   usagepage1
---vid  4013  pid  3151  usage  202   usagepageff55
===

It's works fine on Windows with the  same code  . But can't open device on Mac(Can find  but can not open)

dongjian, 2022-06-24:
It's strange . after some test . I found that I can open the device use "sudo cargo run"

timfish, 2022-06-24:
Some platforms don't let you access keyboard/mouse HID devices as a security precaution. I think this is the expected behaviour on macOS.

closed 2022-06-24 by ruabmbua

dongjian, 2022-06-24:
Thanks!
Links may be helpful to others who have similar problems:
https://github.com/libusb/hidapi/issues/344
https://github.com/libusb/hidapi/pull/397