ruabmbua/hidapi-rs #87: Having issues opening game controller with open_path

LuggaFiech, 2022-07-21:
I just can't get open_path to work. Below is my code. Please let me know what i'm doing wrong here.
[image]

ruabmbua, 2022-07-21:
I assume you are using windows (the path). Can you try opening it with vid + pid instead? I never tried to open by path on windows.

LuggaFiech, 2022-07-21:
My problem is that i need to open only one page of the device. How would i do that only with vid and pid?

ruabmbua, 2022-07-21:
Hm, it might not be possible then. You `ll have to check if the hidapi C library allows that on windows, we use it as the backend.

LuggaFiech, 2022-07-21:
Ok thanks for the answer.

closed 2022-07-21 by LuggaFiech

ruabmbua, 2022-07-21:
You can try listing the devices with their usage pages. If the multiple usage pages do not show up, it might be unimplemented for windows.

LuggaFiech, 2022-07-21:
If i try it for every device it only prints one usage page each time so it might be unimplemented then.

ruabmbua, 2022-07-21:
Only recently this started working on linux hidraw, so it might just be too new. Also, entirely possible that the windows API does not allow it?

LuggaFiech, 2022-07-21:
I assume if you open a device with vid and pid it opens the standard usage page?

ruabmbua, 2022-07-21:
Probably (it does for my devices), but entirely possible it just opens the one first in the descriptors (is that the default).

LuggaFiech, 2022-07-21:
Because i'm building the device myself i have the ability to manage the order of the usage pages so i will try to manage it like that. Thanks for helping.