ruabmbua/hidapi-rs #88: Incorrect Function Error when using get_feature_report

Manul-Keyboard, 2022-07-27:
Code:
[image]
Error:
[image]

ruabmbua, 2022-07-28:
There is not any space in your buffer to actually write the received report into.
Use a larger buf.

Also as per documentation:

> Set the first byte of buf to the ‘Report ID’ of the report to be read. Upon return, the first byte will still contain the Report ID, and the report data will start in buf[1].

The first byte in the given buf is input, and contains the report id you wish to receive.

Manul-Keyboard, 2022-07-29:
Thank you for your answer. It's working now.

closed 2022-07-29 by Manul-Keyboard