ruabmbua/hidapi-rs #89: List Devices shows more than one of the same device

SergioRibera, 2022-08-13:
This causes that when selecting the first or the last one to open, it does not work, so how can I know which one is the correct one or what am I doing wrong?

[image]

I leave you the udev rules that I wrote, maybe they have something to do, even when listing with libusb, only one device is shown

[image]

> /etc/udev/rules.d/99-hid.rules
  ```
# HIDAPI/libusb
SUBSYSTEMS=="usb", ATTRS{idVendor}=="4653", ATTRS{idProduct}=="4d4d", TAG+="uaccess", RUN{builtin}+="uaccess"

# HIDAPI/hidraw
KERNEL=="hidraw*", ATTRS{idVendor}=="4653", ATTRS{idProduct}=="4d4d", TAG+="uaccess"
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="4653", ATTRS{idProduct}=="4d4d", TAG+="uaccess", RUN{builtin}+="uaccess"
  ```

ruabmbua, 2022-08-13:
What do you mean with "does not work"?
I am also not sure what your crkbd_gui command does, when listing, I am interested in what the numbers are exactly.

I suspect the device just has multiple interfaces or usage pages, and hidapi lists them separately. Just one of them will be the "device" you actually want to open.

SergioRibera, 2022-08-13:
Oh right, I forgot to show you that part of the code, it lists all the devices, when I realized that there wasn't one, so I changed the find to a filter
[image]

ruabmbua, 2022-08-13:
The path seems to be different for all the four devices, the last .X is different.
That means there are different interfaces on the device. You have to open the correct one.

I suppose you application will need some code for matching devices, and opening the correct one (with the correct interface id).

ruabmbua, 2022-08-13:
libusb backend does not expose all the interfaces, which is actually a limitation of the implementation.
hidraw backend gives you full access.

ruabmbua, 2022-08-13:
I suggest you make a filter, in which you match for pre-programmed vid + pid + interface number.

Or just create special handling for just this vid + pid. Idk if it is usual for the device kind you handle to have multiple interfaces.

SergioRibera, 2022-08-13:
Ooooooh, using the interface number, it works perfect, now I will see if it works correctly on other platforms, thank you very much.

closed 2022-08-13 by SergioRibera