ruabmbua/hidapi-rs #91: linux-static-hidraw feature flag stops exclusive access of devices

rnd-ash, 2022-09-05:
# Background
I am using this library to create a barcode scanner logger. The HIDAPI is used to read barcode scanner input packets (The scanners emulate a USB keyboard, so the packets are the same as a normal keyboard). The program is designed to take ownership of any scanner plugged in, so the scanners output can be logged by my program in the background rather than sent to whatever window the user is in. 

## Affected versions
Confirmed to affect every version of this library from  1.3.0 - 1.4.2 (Have not tested earlier versions).

## The problem
Due to wanting hotplug support, I needed to have hidapi error implemented to detect when a device was unplugged (Without just throwing `hidapi error: hid_error is not implemented yet`). So I followed the wiki and enabled the `linux-static-hidraw` feature flag. 

Unfortunately, in doing so, my program no longer has exclusive access over the HID scanners it is using. For some reason, when enabling this feature flag, the barcode scanner output is sent to my logger program, but also sent to whatever window the user is hovering over. 

## Screenshot

This image shows the output of the scanner being grabbed by both my barcode-scanner program and an external text editor that my mouse is hovering over. The correct outcome would be that the scanners output is only processed by the scanner program (Which happens when disabling `linux-static-hidraw`)
[image]

ruabmbua, 2022-09-08:
This is totally normal behavior, and how it is supposed to work on linux. The input driver of the system recognizes the hid device, and handles it like it should be. hidapi access the hidraw device also created by the kernel, to provide raw access, it does however not affect the rest of the input system.

To properly solve this problem, you should probably just blacklist the device in X11 / wayland / libinput.

You could try to find out the xinput number of the device and then`xinput --disable NUM` when your program starts and opens the device.

I am sure there is a method for wayland too (via libinput), but I do not know how from the top of my head.

ruabmbua, 2022-11-19:
Will close because of inactivity.

closed 2022-11-19 by ruabmbua