ruabmbua/hidapi-rs #94: linux-static-libusb exports non-functional DeviceInfo::usage and usage_page

micolous, 2022-11-19:
`linux-static-libusb` lacks functionality that other platforms have, such as `DeviceInfo::usage()` and `DeviceInfo::usage_page()`. It appears `libusb` doesn't support this at all, [and hidapi works around it with a rather invasive detach/reattach process](https://github.com/libusb/hidapi/blob/76108294092c023a4ece99eb3219559cea0d5066/libusb/hid.c#L638-L652), and [the documentation is difficult to find](https://github.com/libusb/hidapi/blob/2b70208feabeb4cb524dff58c450af63544790bc/hidapi/hidapi.h#L159-L164).

There are no warnings in `hidapi-rs`' documentation that this isn't supported with the default `linux-static-libusb`, which led to a frustrating troubleshooting process trying to "fix" something that worked fine on macOS and Windows. 🙃 

## Expected behaviour

* `DeviceInfo::usage` and `DeviceInfo::usage_page` should be **entirely removed** when using `libusb` backends on Linux
* `linux-static-hidraw` should be seriously considered as the default backend on Linux (where this functionality actually works)

ruabmbua, 2022-11-19:
I have already considered moving to hidraw by default for many times, each time however, there was some problem with old linux kernel. 

However since the thing with usage  & usage_page is pretty annoying (and even if in docs I doubt everyone will read them), I think its finally time to move to the hidraw backend by default. That will also make it fine to configure out the usage and usage_page properties.

If some projects want to support old kernels, they can switch to the libusb backend, and remove usage & usage_page parts in code.

ruabmbua, 2022-11-19:
Should be fixed now, I published a 2.0.0 version of the crate because of the breaking change.
Also took the time to remove all the deprecated items from the code.

closed 2022-11-19 by ruabmbua

micolous, 2022-11-20:
Thanks for the prompt response!

PS: While `hidraw` is now the default, the module documentation still says `libusb` is the default on the `master` branch at the time of writing: 

https://github.com/ruabmbua/hidapi-rs/blob/fcaf6df669346f067f5c847998499305e4cc3ba1/src/lib.rs#L40-L41

ruabmbua, 2022-11-20:
Thanks for noticing, fixed it and published new crate version so it gets updated on docs.rs