ruabmbua/hidapi-rs #98: Xbox controller not detected and returned by device_list() when plugged in via USB

ludokx, 2023-01-07:
When connecting the controller via BT it's detected using `device_list()` but when it's plugged in via USB, it's not returned at all.

However it's clearly detected on the system (and working) using `lsusb`:
```
Bus 001 Device 045: ID 045e:0b12 Microsoft Corp. Xbox Wireless Controller (model 1914)
```

It works fine with other controllers (DualShock 4, DualSense, Switch Pro).

Any idea what the issue could be or how I could help figure this out?

ruabmbua, 2023-01-08:
I also have a Xbox controller, but not sure how to plug it in via usb. Mine only has bluetooth, and I connect with generic bluetooth on my devices. I can not currently test with it, I am traveling right now.

Maybe there are a couple different models? I remember one, which could only connect to a special microsoft made usb stick via wireless. Do you have this one or a different one?

Anyway, the reason for it not showing up is probably, that the USB protocol used is not HID based. Alternatively it could be HID based, but the implementation of it is buggy, forcing the linux driver that supports it to bypass / quirk the HID stack in the kernel. I think I have heard about xbox one controllers with that problem (mine might be affected). Note, that there is an alternative better driver called xpad-neo for linux, which fixes the problem by creating a virtual hidraw device, where the packets are fixed up.

ruabmbua, 2023-01-08:
You could e.g. look at kernel log or udevadm monitor when plugging in, to see if a hidraw device is created, and if the kernel HID stack is used.

ruabmbua, 2023-01-08:
B.t.w. I really recommend using https://github.com/atar-axis/xpadneo, it even has features that are not supported in windows!

ludokx, 2023-01-09:
> I also have a Xbox controller, but not sure how to plug it in via usb. Mine only has bluetooth, and I connect with generic bluetooth on my devices. I can not currently test with it, I am traveling [...]

I have both a Xbox One S wireless controller and a Xbox Series X wireless controller that I just wire via USB to my SteamDeck dock.

> Anyway, the reason for it not showing up is probably, that the USB protocol used is not HID based. Alternatively it could be HID based, but the implementation of it is buggy, forcing the linux driver that supports it to bypass / quirk the HID stack in the kernel. I think I have heard about xbox one controllers with that problem (mine might be affected). Note, that there is an alternative better driver called https://github.com/atar-axis/xpadneo for linux, which fixes the problem by creating a virtual hidraw device, where the packets are fixed up.

Indeed, checking with `udevadm monitor` it doesn't show a hidraw device getting created. :/

closed 2023-01-22 by ruabmbua