ruabmbua/hidapi-rs #102: Having issues with open_path

Manul-Keyboard, 2023-02-14:
Code:
[image]
Error:
[image]
The error was produced on Linux.

ruabmbua, 2023-02-14:
If you are trying to open a specific interface of a device in a portable way, maybe this is of interest: https://github.com/ruabmbua/hidapi-rs/issues/101

ruabmbua, 2023-02-14:
Note that the *paths* are not portable at all across operating systems, and more of an implementation detail. They are mostly useful, if you somehow already have the path you want to use. The "1-3:1.0" might only be e.g. for windows.

Also I just noticed that the legacy libusb backend has paths like this, but hidapi uses hidraw as the default now.

ruabmbua, 2023-02-14:
Also a small tip: when filing issues, please do not include code snippets as pictures, rather use the proper code comments, or just plain text. Its not super important but it makes life a bit more pleasant for people trying to reproduce issues, because they can just copy the text. Also, for search purposes text is much better.

Manul-Keyboard, 2023-02-14:
Thank you for your answer. I will use code comments in the future.

closed 2023-02-14 by Manul-Keyboard