ruabmbua/hidapi-rs #112: Do i need to write a windows driver to be able to talk to my custom USB HID device?

Aggeloz, 2023-04-28:
Im using an ESP32S2 which has native usb support and i want to be able to send reports between a rust program and my custom usb device but so far the only way i can read data from the device is on linux when i give sudo permissions to the device path. In windows i can open the device but when i send a report the rust program cant read it at all.

lnicola, 2023-04-28:
> so far the only way i can read data from the device is on linux when i give sudo permissions to the device path

Not sure about the Windows question, but on Linux this is usually handled with an udev rule, e.g.:

```
ACTION=="add|change", SUBSYSTEMS=="usb", ATTRS{idVendor}=="04d9", ATTRS{idProduct}=="a052", MODE:="0666"
```

or

```
ACTION=="add|change", SUBSYSTEMS=="usb", ATTRS{idVendor}=="04d9", ATTRS{idProduct}=="a052", GROUP:="uucp"
```

Aggeloz, 2023-04-28:
> If I think about it, that CO2 meter sample used to work fine for me on Windows, so it might be something specific to your device.

Where could i find that code sample?

closed 2023-04-28 by Aggeloz