ruabmbua/hidapi-rs #113: hidapi crate fails to build on FreeBSD

tingox, 2023-05-04:
Trying to install probe-run on FreeBSD, it fails on building hidapi crate
```
tingo@kg-core1:~/personal/projects/2023/rust/embedded/nrf53/nrf5340-app-hal-example $ cargo install probe-run
    Updating crates.io index
  Downloaded probe-run v0.3.8
  Downloaded 1 crate (1.8 MB) in 2.06s
  Installing probe-run v0.3.8
  Downloaded tracing-attributes v0.1.24
  Downloaded flate2 v1.0.26
  Downloaded anstream v0.3.2
  Downloaded signal-hook-registry v1.4.1
  Downloaded probe-rs-target v0.17.0
  Downloaded crc32fast v1.3.2
  Downloaded miniz_oxide v0.7.1
  Downloaded signal-hook v0.3.15
  Downloaded anyhow v1.0.71
  Downloaded cpp_demangle v0.4.1
  Downloaded rustc-demangle v0.1.23
  Downloaded clap v4.2.7
  Downloaded clap_builder v4.2.7
  Downloaded probe-rs v0.17.0
  Downloaded pkg-config v0.3.27
  Downloaded rustix v0.37.19
  Downloaded 16 crates (4.3 MB) in 0.56s (largest was `probe-rs` at 3.3 MB)
   Compiling proc-macro2 v1.0.56
   Compiling unicode-ident v1.0.8
   Compiling quote v1.0.26
   Compiling serde_derive v1.0.160
   Compiling serde v1.0.160
   Compiling libc v0.2.142
   Compiling autocfg v1.1.0
   Compiling cc v1.0.79
   Compiling cfg-if v1.0.0
   Compiling pkg-config v0.3.27
   Compiling syn v2.0.15
   Compiling indexmap v1.9.3
   Compiling io-lifetimes v1.0.10
   Compiling bitflags v1.3.2
   Compiling log v0.4.17
   Compiling crc32fast v1.3.2
   Compiling libusb1-sys v0.6.4
   Compiling adler v1.0.2
   Compiling syn v1.0.109
   Compiling rustix v0.37.19
   Compiling memchr v2.5.0
   Compiling miniz_oxide v0.7.1
   Compiling errno v0.3.1
   Compiling itoa v1.0.6
   Compiling thiserror v1.0.40
   Compiling hashbrown v0.12.3
   Compiling flate2 v1.0.26
   Compiling hidapi v2.2.2
   Compiling num-traits v0.2.15
   Compiling once_cell v1.17.1
   Compiling stable_deref_trait v1.2.0
   Compiling unsafe-libyaml v0.2.8
   Compiling anyhow v1.0.71
   Compiling rusb v0.9.2
   Compiling fallible-iterator v0.2.0
   Compiling proc-macro-hack v0.5.20+deprecated
   Compiling radium v0.7.0
   Compiling ryu v1.0.13
   Compiling utf8parse v0.2.1
   Compiling base64 v0.21.0
   Compiling anstyle-parse v0.2.0
   Compiling thiserror-impl v1.0.40
   Compiling gimli v0.27.2
   Compiling object v0.30.3
   Compiling is-terminal v0.4.7
   Compiling anstyle v1.0.0
[... 51 lines cut ...]
  |
  = note: LC_ALL="C" PATH="/usr/home/tingo/.rustup/toolchains/stable-x86_64-unknown-freebsd/lib/rustlib/x86_64-unknown-freebsd/bin:/home/tingo/.cargo/bin:/sbin:/bin:/usr/sbin:/usr/bin:/usr/local/sbin:/usr/local/bin:/home/tingo/bin:/home/tingo/.local/bin" VSLANG="1033" "cc" "-m64" "/tmp/rustcWjS9e9/symbols.o" "/tmp/cargo-installRJVseR/release/deps/probe_run-557d11e13cbd19bd.probe_run.c0fb2077-cgu.0.rcgu.o" "/tmp/cargo-installRJVseR/release/deps/probe_run-557d11e13cbd19bd.probe_run.c0fb2077-cgu.1.rcgu.o" "/tmp/cargo-installRJVseR/release/deps/probe_run-557d11e13cbd19bd.probe_run.c0fb2077-cgu.10.rcgu.o" "/tmp/cargo-installRJVseR/release/deps/probe_run-557d11e13cbd19bd.probe_run.c0fb2077-cgu.11.rcgu.o" "/tmp/cargo-installRJVseR/release/deps/probe_run-557d11e13cbd19bd.probe_run.c0fb2077-cgu.12.rcgu.o" "/tmp/cargo-installRJVseR/release/deps/probe_run-557d11e13cbd19bd.probe_run.c0fb2077-cgu.13.rcgu.o" "/tmp/cargo-installRJVseR/release/deps/probe_run-557d11e13cbd19bd.probe_run.c0fb2077-cgu.14.rcgu.o" "/tmp/cargo-installRJVseR/release/deps/probe_run-557d11e13cbd19bd.probe_run.c0fb2077-cgu.15.rcgu.o" "/tmp/cargo-installRJVseR/release/deps/probe_run-557d11e13cbd19bd.probe_run.c0fb2077-cgu.2.rcgu.o" "/tmp/cargo-installRJVseR/release/deps/probe_run-557d11e13cbd19bd.probe_run.c0fb2077-cgu.3.rcgu.o" "/tmp/cargo-installRJVseR/release/deps/probe_run-557d11e13cbd19bd.probe_run.c0fb2077-cgu.4.rcgu.o" "/tmp/cargo-installRJVseR/release/deps/probe_run-557d11e13cbd19bd.probe_run.c0fb2077-cgu.5.rcgu.o" "/tmp/cargo-installRJVseR/release/deps/probe_run-557d11e13cbd19bd.probe_run.c0fb2077-cgu.6.rcgu.o" "/tmp/cargo-installRJVseR/release/deps/probe_run-557d11e13cbd19bd.probe_run.c0fb2077-cgu.7.rcgu.o" "/tmp/cargo-installRJVseR/release/deps/probe_run-557d11e13cbd19bd.probe_run.c0fb2077-cgu.8.rcgu.o" "/tmp/cargo-installRJVseR/release/deps/probe_run-557d11e13cbd19bd.probe_run.c0fb2077-cgu.9.rcgu.o" "/tmp/cargo-installRJVseR/release/deps/probe_run-557d11e13cbd19bd.324wiexwmv6v8awb.rcgu.o" "-Wl,--as-needed" "-L" "/tmp/cargo-installRJVseR/release/deps" "-L" "/usr/local/lib" "-L" "/usr/lib" "-L" "/usr/home/tingo/.rustup/toolchains/stable-x86_64-unknown-freebsd/lib/rustlib/x86_64-unknown-freebsd/lib" "-Wl,-Bstatic" "/tmp/cargo-installRJVseR/release/deps/libgit_version-800ea17d5a2057c1.rlib" "/tmp/cargo-installRJVseR/release/deps/libdefmt_decoder-1c384c4918cd3703.rlib" "/tmp/cargo-installRJVseR/release/deps/libserde_json-2814fdaadf688b5e.rlib" "/tmp/cargo-installRJVseR/release/deps/libdissimilar-5aa7096232ddc703.rlib" "/tmp/cargo-installRJVseR/release/deps/libdefmt_json_schema-f24ac01520259221.rlib" "/tmp/cargo-installRJVseR/release/deps/libtime-2de3880115d8e13d.rlib" "/tmp/cargo-installRJVseR/release/deps/libtime_core-7f6462ec92c4a4e7.rlib" "/tmp/cargo-installRJVseR/release/deps/libdefmt_parser-30e67e641261e49c.rlib" "/tmp/cargo-installRJVseR/release/deps/libclap-d89398331f8705ba.rlib" "/tmp/cargo-installRJVseR/release/deps/libclap_builder-d9e1068a83cc282a.rlib" "/tmp/cargo-installRJVseR/release/deps/libstrsim-4bee60204ac3ec40.rlib" "/tmp/cargo-installRJVseR/release/deps/libanstream-389843148cc84fb6.rlib" "/tmp/cargo-installRJVseR/release/deps/libanstyle_query-bc327a69da9fd9a6.rlib" "/tmp/cargo-installRJVseR/release/deps/libis_terminal-95cebc877ee8f067.rlib" "/tmp/cargo-installRJVseR/release/deps/librustix-a1c333fdfcb05290.rlib" "/tmp/cargo-installRJVseR/release/deps/liberrno-01da4d81480681d0.rlib" "/tmp/cargo-installRJVseR/release/deps/libio_lifetimes-c55cefe5bff9923f.rlib" "/tmp/cargo-installRJVseR/release/deps/libanstyle-872319d473c5fd18.rlib" "/tmp/cargo-installRJVseR/release/deps/libcolorchoice-869a6e0868afaa2b.rlib" "/tmp/cargo-installRJVseR/release/deps/libanstyle_parse-9125c9c7b8d86fc5.rlib" "/tmp/cargo-installRJVseR/release/deps/libutf8parse-f7017607055ecaaa.rlib" "/tmp/cargo-installRJVseR/release/deps/libclap_lex-1cf01a2e5487e037.rlib" "/tmp/cargo-installRJVseR/release/deps/libaddr2line-b7520e973821daf5.rlib" "/tmp/cargo-installRJVseR/release/deps/librustc_demangle-2c6752d8d3bab9fb.rlib" "/tmp/cargo-installRJVseR/release/deps/libcpp_demangle-ab49e3397d640c92.rlib" "/tmp/cargo-installRJVseR/release/deps/libcolored-c785fa25a1a6f40c.rlib" "/tmp/cargo-installRJVseR/release/deps/liblazy_static-20565780dedb67dc.rlib" "/tmp/cargo-installRJVseR/release/deps/libatty-0a494793a1aee318.rlib" "/tmp/cargo-installRJVseR/release/deps/libsignal_hook-4f533a2f4bb5d0f1.rlib" "/tmp/cargo-installRJVseR/release/deps/libsignal_hook_registry-f0ed217cc075e0b3.rlib" "/tmp/cargo-installRJVseR/release/deps/libprobe_rs-9edfa6b11e93a1d9.rlib" "/tmp/cargo-installRJVseR/release/deps/libkmp-6213e75043113001.rlib" "/tmp/cargo-installRJVseR/release/deps/libbincode-73f3795846e35527.rlib" "/tmp/cargo-installRJVseR/release/deps/libserde_yaml-e0a5286de5a41b5e.rlib" "/tmp/cargo-installRJVseR/release/deps/libryu-faa0850cce6829a2.rlib" "/tmp/cargo-installRJVseR/release/deps/libitoa-67875edd3f1d91b6.rlib" "/tmp/cargo-installRJVseR/release/deps/libindexmap-f6ec650ac4d17118.rlib" "/tmp/cargo-installRJVseR/release/deps/libhashbrown-0ca316ef8ab2c06a.rlib" "/tmp/cargo-installRJVseR/release/deps/libunsafe_libyaml-ffbd2658247c3a14.rlib" "/tmp/cargo-installRJVseR/release/deps/libstatic_assertions-ad077abc31ca0f49.rlib" "/tmp/cargo-installRJVseR/release/deps/libthiserror-05ff6f50cc967b4b.rlib" "/tmp/cargo-installRJVseR/release/deps/libjaylink-7cf23dee53e32ff9.rlib" "/tmp/cargo-installRJVseR/release/deps/libbyteorder-49b6944364891cd5.rlib" "/tmp/cargo-installRJVseR/release/deps/libbitflags-8479ec16f9e8b912.rlib" "/tmp/cargo-installRJVseR/release/deps/librusb-310f195d5a8599a7.rlib" "/tmp/cargo-installRJVseR/release/deps/liblibusb1_sys-83f511982f91187a.rlib" "/tmp/cargo-installRJVseR/release/deps/libhidapi-89e8cf709d666757.rlib" "/tmp/cargo-installRJVseR/release/deps/liblibc-1f3d9859399ed516.rlib" "/tmp/cargo-installRJVseR/release/deps/libscroll-9b28af9fb613086d.rlib" "/tmp/cargo-installRJVseR/release/deps/libsvg-4c367842d0f3bfae.rlib" "/tmp/cargo-installRJVseR/release/deps/libihex-6423c49f0698f2be.rlib" "/tmp/cargo-installRJVseR/release/deps/libtracing-40c99f427f29e084.rlib" "/tmp/cargo-installRJVseR/release/deps/liblog-7b6f303958628c85.rlib" "/tmp/ca [... cut ...]
13.2-RELEASE
tingo@kg-core1:~/personal/projects/2023/rust/embedded/nrf53/nrf5340-app-hal-example $ uname -a
FreeBSD kg-core1.kg4.no 13.2-RELEASE FreeBSD 13.2-RELEASE releng/13.2-n254617-525ecfdad597 GENERIC amd64
```
with latest hidapi installed
```
tingo@kg-core1:~/personal/projects/2023/rust/embedded/nrf53/nrf5340-app-hal-example $ pkg info hid\*
hidapi-0.13.1
```
I'm very new to rust so I don't know any more info to add now. If there is any more info I can add to debug this, please let me know.

ruabmbua, 2023-05-06:
@neverpanic implemented freebsd support for  this library in commit 0a2753067d9bd9a25fcec3c7d29ebb649c928282.
Unfortunately I myself have 0 experience working with FreeBSD (apart from OPNSense), but the error looks like it can not find libusb functions.

The fix might be as simple as adding `pkg_config::probe_library("libusb").expect("Unable to find hidapi");` to the build.rs of hidapi.

Alternatively we can avoid using libusb_set_option on freebsd since its probably not needed anyway.

ruabmbua, 2023-05-06:
Note to myself: I really need to setup CI again...

We had multiple CI setups already, but all of them removed free tier / had some other problems.

closed 2023-05-06 by ruabmbua

ruabmbua, 2023-05-06:
https://github.com/probe-rs/probe-rs/pull/1606