ruabmbua/hidapi-rs #128: `linux-native` feature does not compile

ShayBox, 2023-07-20:
Enabling the `linux-native` feature whether or not I also disable default features causes the following errors

```
error[E0463]: can't find crate for `nix`
  --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\lib.rs:62:1
   |
62 | extern crate nix;
   | ^^^^^^^^^^^^^^^^^ can't find crate

error[E0252]: the name `HidApiBackend` is defined multiple times
  --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\lib.rs:97:5
   |
95 | use crate::hidapi::HidApiBackend;
   |     ---------------------------- previous import of the type `HidApiBackend` here
96 | #[cfg(feature = "linux-native")]
97 | use linux_native::HidApiBackend;
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^ `HidApiBackend` reimported here
   |
   = note: `HidApiBackend` must be defined only once in the type namespace of this module
help: you can use `as` to change the binding name of the import
   |
97 | use linux_native::HidApiBackend as OtherHidApiBackend;
   |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

error[E0433]: failed to resolve: could not find `unix` in `os`
  --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\linux_native.rs:12:9
   |
12 |         unix::{ffi::OsStringExt, fs::OpenOptionsExt},
   |         ^^^^ could not find `unix` in `os`

error[E0432]: unresolved import `std::os::fd`
  --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\linux_native.rs:11:9
   |
11 |         fd::{AsRawFd, OwnedFd},
   |         ^^ could not find `fd` in `os`

error[E0432]: unresolved imports `ioctl::hidraw_ioc_get_feature`, `ioctl::hidraw_ioc_grdescsize`, `ioctl::hidraw_ioc_set_feature`
  --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\linux_native.rs:25:13
   |
25 | use ioctl::{hidraw_ioc_get_feature, hidraw_ioc_grdescsize, hidraw_ioc_set_feature};
   |             ^^^^^^^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^^^^^^^^ no `hidraw_ioc_set_feature` in `linux_native::ioctl`
   |             |                       |
   |             |                       no `hidraw_ioc_grdescsize` in `linux_native::ioctl`
   |             |                       help: a similar name exists in the module: `HIDRAW_IOC_GRDESCSIZE`
   |             no `hidraw_ioc_get_feature` in `linux_native::ioctl`

error: cannot determine resolution for the macro `ioctl_read`
  --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\linux_native\ioctl.rs:11:1      
   |
11 | ioctl_read!(
   | ^^^^^^^^^^
   |
   = note: import resolution is stuck, try simplifying macro imports

error: cannot determine resolution for the macro `ioctl_readwrite_buf`
  --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\linux_native\ioctl.rs:18:1      
   |
18 | ioctl_readwrite_buf!(
   | ^^^^^^^^^^^^^^^^^^^
   |
   = note: import resolution is stuck, try simplifying macro imports

error: cannot determine resolution for the macro `ioctl_readwrite_buf`
  --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\linux_native\ioctl.rs:24:1      
   |
24 | ioctl_readwrite_buf!(
   | ^^^^^^^^^^^^^^^^^^^
   |
   = note: import resolution is stuck, try simplifying macro imports

error[E0433]: failed to resolve: use of undeclared crate or module `udev`
  --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\linux_native.rs:39:36
   |
39 |         let mut enumerator = match udev::Enumerator::new() {
   |                                    ^^^^ use of undeclared crate or module `udev`

error[E0433]: failed to resolve: use of undeclared crate or module `udev`
   --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\linux_native.rs:586:22
    |
586 |         let device = udev::Device::from_syspath(&syspath)?;
    |                      ^^^^ use of undeclared crate or module `udev`

error[E0425]: cannot find value `O_CLOEXEC` in crate `libc`
   --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\linux_native.rs:438:33
    |
438 |             .custom_flags(libc::O_CLOEXEC | libc::O_NONBLOCK)
    |                                 ^^^^^^^^^ not found in `libc`

error[E0425]: cannot find value `O_NONBLOCK` in crate `libc`
   --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\linux_native.rs:438:51
    |
438 |             .custom_flags(libc::O_CLOEXEC | libc::O_NONBLOCK)
    |                                                   ^^^^^^^^^^ not found in `libc`

error[E0046]: not all trait items implemented, missing: `check_error`
   --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\linux_native.rs:474:1
    |
474 | impl HidDeviceBackendBase for HidDevice {
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ missing `check_error` in implementation
    |
   ::: C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\lib.rs:427:5
    |
427 |     fn check_error(&self) -> HidResult<HidError>;
    |     --------------------------------------------- `check_error` from trait

error[E0599]: no method named `custom_flags` found for mutable reference `&mut OpenOptions` in the current scope     
   --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\linux_native.rs:438:14
    |
435 |           let fd: OwnedFd = match OpenOptions::new()
    |  _________________________________-
436 | |             .read(true)
437 | |             .write(true)
438 | |             .custom_flags(libc::O_CLOEXEC | libc::O_NONBLOCK)
    | |             -^^^^^^^^^^^^ method not found in `&mut OpenOptions`
    | |_____________|
    | 
    |
    = help: items from traits can only be used if the trait is in scope
help: the following trait is implemented but not in scope; perhaps add a `use` for it:
    |
3   + use std::os::windows::fs::OpenOptionsExt;
    |

error[E0433]: failed to resolve: use of undeclared crate or module `udev`
  --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\linux_native.rs:70:43
   |
70 | fn device_to_hid_device_info(raw_device: &udev::Device) -> Option<Vec<DeviceInfo>> {
   |                                           ^^^^ use of undeclared crate or module `udev`

error[E0433]: failed to resolve: use of undeclared crate or module `udev`
   --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\linux_native.rs:168:25
    |
168 | fn fill_in_usb(device: &udev::Device, info: DeviceInfo, name: &OsStr) -> DeviceInfo {
    |                         ^^^^ use of undeclared crate or module `udev`

error[E0433]: failed to resolve: use of undeclared crate or module `udev`
   --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\linux_native.rs:355:29
    |
355 | fn attribute_as_wchar(dev: &udev::Device, attr: &str) -> WcharString {
    |                             ^^^^ use of undeclared crate or module `udev`

error[E0433]: failed to resolve: use of undeclared crate or module `udev`
   --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\linux_native.rs:365:27
    |
365 | fn attribute_as_i32(dev: &udev::Device, attr: &str) -> Option<i32> {
error[E0433]: failed to resolve: use of undeclared crate or module `udev`
   --> C:\Users\Admin\.cargo\registry\src\github.com-1ecc6299db9ec823\hidapi-2.4.0\src\linux_native.rs:374:27
    |
374 | fn attribute_as_u16(dev: &udev::Device, attr: &str) -> Option<u16> {
    |                           ^^^^ use of undeclared crate or module `udev`

Some errors have detailed explanations: E0046, E0252, E0425, E0432, E0433, E0463, E0599.
For more information about an error, try `rustc --explain E0046`.
error: could not compile `hidapi` (lib) due to 19 previous errors
warning: build failed, waiting for other jobs to finish...
```

ruabmbua, 2023-07-20:
Can you provide an example? I can not really reproduce the issue.

ShayBox, 2023-07-20:
I just added hidapi to my project with Linux-native enabled, there's no code yet. It's just failing to compile

ruabmbua, 2023-07-21:
A thx, I am actually aware of this, forgot to cherry pick it from the current native windows implementation branch.
I will do itsometime today.

closed 2023-07-21 by ruabmbua