ruabmbua/hidapi-rs #132: error: could not find system library 'libudev' required by the 'hidapi' crate

0xjinn, 2023-08-24:
Hi team,

When I try to build my rust project in github CI (Under linux machine), I get the following error:

Even though, I already have `apt-get install libudev-dev` ran before cargo build. 

Any idea?

```
#74 6.588 error: failed to run custom build command for `hidapi v1.5.0`
#74 6.590 
#74 6.590 Caused by:
#74 6.590   process didn't exit successfully: `/aptos/target/default/release/build/hidapi-6ba6a7d34105e233/build-script-build` (exit status: 101)
#74 6.590   --- stdout
#74 6.590   cargo:rerun-if-env-changed=LIBUDEV_NO_PKG_CONFIG
#74 6.590   cargo:rerun-if-env-changed=PKG_CONFIG_x86_64-unknown-linux-gnu
#74 6.590   cargo:rerun-if-env-changed=PKG_CONFIG_x86_64_unknown_linux_gnu
#74 6.590   cargo:rerun-if-env-changed=HOST_PKG_CONFIG
#74 6.590   cargo:rerun-if-env-changed=PKG_CONFIG
#74 6.590   cargo:rerun-if-env-changed=LIBUDEV_STATIC
#74 6.590   cargo:rerun-if-env-changed=LIBUDEV_DYNAMIC
#74 6.590   cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
#74 6.590   cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
#74 6.590   cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
#74 6.590   cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
#74 6.590   cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
#74 6.590   cargo:rerun-if-env-changed=PKG_CONFIG_PATH
#74 6.590   cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
#74 6.590   cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
#74 6.590   cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
#74 6.590   cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
#74 6.590   cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
#74 6.590   cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
#74 6.590   cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
#74 6.590   cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR
#74 6.590 
#74 6.590   --- stderr
#74 6.590   thread 'main' panicked at 'Unable to find libudev: `PKG_CONFIG_ALLOW_SYSTEM_CFLAGS="1" PKG_CONFIG_ALLOW_SYSTEM_LIBS="1" "pkg-config" "--libs" "--cflags" "libudev"` did not exit successfully: exit status: 1
#74 6.590   error: could not find system library 'libudev' required by the 'hidapi' crate
#74 6.590 
#74 6.590   --- stderr
#74 6.590   Package libudev was not found in the pkg-config search path.
#74 6.590   Perhaps you should add the directory containing `libudev.pc'
#74 6.590   to the PKG_CONFIG_PATH environment variable
#74 6.590   No package 'libudev' found
#74 6.590   ', /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/hidapi-1.5.0/build.rs:57:54
#74 6.590   note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
#74 6.590 warning: build failed, waiting for other jobs to finish...
#74 ERROR: process "/bin/sh -c docker/builder/build-tools.sh" did not complete successfully: exit code: 101

```

lnicola, 2023-08-24:
So what does `PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 pkg-config --libs --cflags libudev` output in your container? In mine, it says `-I/usr/include -L/usr/lib/x86_64-linux-gnu -ludev`.

0xjinn, 2023-08-24:
> So what does `PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 pkg-config --libs --cflags libudev` output in your container? In mine, it says `-I/usr/include -L/usr/lib/x86_64-linux-gnu -ludev`.

Tried it on the CI container, it shows the following:

```
#59 1.468 Package libudev was not found in the pkg-config search path.
#59 1.468 Perhaps you should add the directory containing `libudev.pc'
#59 1.468 to the PKG_CONFIG_PATH environment variable
#59 1.468 No package 'libudev' found
```

ruabmbua, 2023-08-24:
In our CI we do this: https://github.com/ruabmbua/hidapi-rs/blob/78f273800a7e9fc7f428e45d2cfebaf7b44862c1/.github/workflows/rust.yml#L34

ruabmbua, 2023-08-24:
and for linux-native: https://github.com/ruabmbua/hidapi-rs/blob/78f273800a7e9fc7f428e45d2cfebaf7b44862c1/.github/workflows/rust.yml#L69C29-L69C44

lnicola, 2023-08-24:
/usr/local/cargo/registry/src/ looks a little non-standard, maybe you can try compiling just hidapi in a plain image (we're using Ubuntu 22.04 here), then see what's different in your environment.

ruabmbua, 2023-08-24:
Testing with PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 pkg-config --libs --cflags is probably better. As long as it does not return that it found the lib, it can`t work.

0xjinn, 2023-08-24:
Hey guys, thank you for helping!

I think I just found the issue. Its because that libudev and pkg-config seems to be installed using different package-manager. After I explicitly enforcing both to install using apt-get. Cargo build works again! 

Closing this issue now.

closed 2023-08-24 by 0xjinn