ruabmbua/hidapi-rs #146: Rename branch `master` to `main`

uklotzde, 2024-01-28:
- Shorter
- More common now
- Inoffensive

I guess GitHub even offers you this option in the repo settings to perform this migration.

ruabmbua, 2024-01-28:
* Well I agree that its shorter (although I do not think the small difference is important).
* Not really more common than master for most repos I work with (although most of them are not open source).
* Inoffensive: I get that this seems to be very important to some people. I personally do not really care that much, I never before associated the word with slavery in that context (especially since *slave* is not a common branch name / name in git also).

Because I somewhat care that most repos I work with / have ownership over have the same *main* branch name for convenience, I am going to ACK this request, and change the name. Changing every  other repository that currently uses *main* over to the name *master* would probably be a bit controversial :sweat_smile: .

ruabmbua, 2024-01-28:
Also sorry to anyone who has to now update their local branch refs.

closed 2024-01-28 by ruabmbua

uklotzde, 2024-01-28:
Shorter and the new default are the major motivations for updating existing repos. I also don't really care about the meaning of words in other, unrelated contexts.

Everyone who relied on the default branch for Git dependencies will not notice the change. If you depend on the repo and a particular branch instead of released versions you should be prepared for breaking changes anyway.

Sometimes breaking changes are inevitable to improve consistency. It pays out in the long term.