ruabmbua/hidapi-rs #149: Creating GitHub releases for new versions

timrogers, 2024-02-05:
Hey there 👋

Thank you for this great package - I'm using it in one of my open source projects, https://github.com/timrogers/litra-rs, and it's really easy to use 🥳 

But I'd like to make a request - would you be willing to create a GitHub release when you release new versions of the package?

At the moment, it's difficult to review new versions and understand what has changed. If you do this, then Dependabot PRs [like this one](https://github.com/timrogers/litra-rs/pull/12) will automatically explain changes.

Thanks again!

Tim

ruabmbua, 2024-02-06:
I will keep the ticket open until the next release, then hopefully close it. You can reopen it if I forget again :smile: .

Generally we try to hold to semver, so unless there is a major version change you should be able to just upgrade, and it should keep working.

ruabmbua, 2024-02-07:
There is already a new release!

closed 2024-02-07 by ruabmbua