ruabmbua/hidapi-rs #153: warning: hidapi@2.6.1: cc1: fatal error: etc/hidapi/linux/hid.c: No such file or directory

ChocolateLoverRaj, 2024-04-06:
I get this error when building anything that has uses this Rust library. I get this error on NixOS and in a Debian 12 VM.
Steps to reproduce:
- Create a Debian 12 vm
- Install Rust
- Run `sudo apt install libhidapi-dev`
- Clone the repo and enter the dir
- Run `cargo run --example lshid`
Full error:
```   Compiling hidapi v2.6.1 (/home/test/hidapi-rs)
The following warnings were emitted during compilation:

warning: hidapi@2.6.1: Compiler version doesn't include clang or GCC: "cc" "--version"
warning: hidapi@2.6.1: cc1: fatal error: etc/hidapi/linux/hid.c: No such file or directory
warning: hidapi@2.6.1: compilation terminated.

error: failed to run custom build command for `hidapi v2.6.1 (/home/test/hidapi-rs)`

Caused by:
  process didn't exit successfully: `/home/test/hidapi-rs/target/debug/build/hidapi-aa41589f8819b5ac/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-changed=etc/hidapi/linux/hid.c
  cargo:rerun-if-env-changed=LIBUDEV_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=LIBUDEV_STATIC
  cargo:rerun-if-env-changed=LIBUDEV_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=SYSROOT
  cargo:rerun-if-env-changed=LIBUDEV_STATIC
  cargo:rerun-if-env-changed=LIBUDEV_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rustc-link-search=native=/usr/lib/x86_64-linux-gnu
  cargo:rustc-link-lib=udev
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=LIBUDEV_STATIC
  cargo:rerun-if-env-changed=LIBUDEV_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR
  TARGET = Some("x86_64-unknown-linux-gnu")
  OPT_LEVEL = Some("0")
  HOST = Some("x86_64-unknown-linux-gnu")
  cargo:rerun-if-env-changed=CC_x86_64-unknown-linux-gnu
  CC_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CC_x86_64_unknown_linux_gnu
  CC_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_CC
  HOST_CC = None
  cargo:rerun-if-env-changed=CC
  CC = None
  cargo:rerun-if-env-changed=CC_ENABLE_DEBUG_OUTPUT
  cargo:warning=Compiler version doesn't include clang or GCC: "cc" "--version"
  cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some("true")
  CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
  cargo:rerun-if-env-changed=CFLAGS_x86_64-unknown-linux-gnu
  CFLAGS_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CFLAGS_x86_64_unknown_linux_gnu
  CFLAGS_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_CFLAGS
  HOST_CFLAGS = None
  cargo:rerun-if-env-changed=CFLAGS
  CFLAGS = None
  cargo:warning=cc1: fatal error: etc/hidapi/linux/hid.c: No such file or directory
  cargo:warning=compilation terminated.

  --- stderr

  error occurred: Command "cc" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-gdwarf-4" "-fno-omit-frame-pointer" "-m64" "-I" "etc/hidapi/hidapi" "-Wall" "-Wextra" "-o" "/home/test/hidapi-rs/target/debug/build/hidapi-851155bf1a844971/out/e6d9cf3cdc9c3cd3-hid.o" "-c" "etc/hidapi/linux/hid.c" with args "cc" did not execute successfully (status code exit status: 1)
```

ruabmbua, 2024-04-06:
I am guessing you just forgot to initialize the git submodules?

This rust library is a wrapper around libusb/hidapi, and by default it builds the C library from source to statically link it. For this it expects a git submodule at etc/hidapi.
You can initialize the submodule by running `git submodule update --init`.

Alternatively you can use the newer rust only backend by compiling with --no-default-features --features=linux-native. It does not have any dependency any more to the hidapi C library. Not the default yet, but should work out fine.

ChocolateLoverRaj, 2024-04-07:
> I am guessing you just forgot to initialize the git submodules?
Yes

After downloading the submodules the example works. How do I do it when using this library in another Rust project? Do I have to create a submodule for that project in a `etc` folder too?

ChocolateLoverRaj, 2024-04-07:
Also I'll probably use the `linux-native` feature because it's an easy fix and Rust is awesome and C is not.

ruabmbua, 2024-04-07:
> > I am guessing you just forgot to initialize the git submodules?
> > Yes
> 
> After downloading the submodules the example works. How do I do it when using this library in another Rust project? Do I have to create a submodule for that project in a `etc` folder too?

If you are using this crate as a dependency via Cargo from the official crates.io release, you do not have to do anything. Cargo automatically packages git submodules when packaging and publishing to crates.io.

closed 2024-04-07 by ruabmbua