ruabmbua/hidapi-rs #160: Installation notes

Resonanz, 2024-11-02:
Could you please add the following install notes? I had to search all over the place to simply get the Rust code to compile.

### Linux (Ubuntu)

After cloning this ```hidapi-rs``` repository  you MUST install ```libusb```: 

```
sudo apt update
sudo apt install libusb-1.0-0-dev
```

This will install the latest stable version of libusb.

Then you MUST then update the ```hidapi-rs``` submodules:

```
git submodule update --init
```

Now the example should run:

```
cargo run --example lshid
```

ruabmbua, 2024-11-02:
> After cloning this hidapi-rs repository you MUST install libusb:

libusb is by default not required on linux, to make this library work. Not for building, and not for running the build program. There are however alternative backends for the library that can use libusb. None of them are enabled by default for linux.

If it does not work for you without libusb, this might be a bug. I just tried it to make sure it works without libusb, and it seems to work for me.

> Then you MUST then update the hidapi-rs submodules:

Fair point, I should probably add this to the README.md in case people clone the library directly instead of using it as
a cargo dependency.

Resonanz, 2024-11-02:
Curious. I use Ubuntu 24.04 and I seemed to require the libusb install.

Seems I also need udev rules to provide permissions.

Resonanz, 2024-11-02:
> Curious. I use Ubuntu 24.04 and I seemed to require the libusb install.
> 
> Seems I also need udev rules to provide permissions. 
> 

ChatGPT.

Does ubuntu 24.04 install libusb by default?

Ubuntu 24.04 LTS does not install libusb by default, but it is available in its repositories. You can easily add it if your applications need it, either by using the apt package manager or by installing via the Ubuntu Software app (App Center).

If you’re setting up for development or certain USB-related tasks, simply run the following command:

sudo apt install libusb-1.0-0-dev

This will install the development version needed for most programming uses, though libusb-1.0-0 should work if you just need basic functionality. The command provides the latest stable libusb libraries for Ubuntu 24.04 LTS.

ruabmbua, 2024-11-02:
Can you maybe try it again after you initialized the submodule? Also the build output when it fails would be appreciated.

> Seems I also need udev rules to provide permissions.

Thats just how this works on linux. If your device is not yet handled by one of the many already shipped rules in your distro, then you have to add your own rule. If you intend to write an app for a particular device, write a rule with its VID / PID and install it as part of the your application.

closed 2024-11-02 by ruabmbua

Resonanz, 2024-11-03:
Thanks for your help. The following worked for me:

I got the udev rules sorted as following (in case you wnt to add them to the README.

Applying ```udev``` rules (thanks to ChatGPT)

Create a new ```udev``` rule file for HID devices. Open a new file in ```/etc/udev/rules.d/``` (e.g., ```99-hidraw-permissions.rules```):

```sudo nano /etc/udev/rules.d/99-hidraw-permissions.rules```

Add a rule to grant access. This example gives members of the plugdev group permission to read and write to hidraw devices:

```KERNEL=="hidraw*", SUBSYSTEM=="hidraw", MODE="0660", GROUP="plugdev"```

Save the file, then reload the ```udev``` rules and trigger them:

```sudo udevadm control --reload-rules```
```sudo udevadm trigger```

Add your user to the ```plugdev``` group:

```sudo usermod -aG plugdev $USER```

Log out and log back in to apply the group change.

After completing these steps, you should be able to access HID devices without needing ```sudo```.

Resonanz, 2024-11-03:
Regarding the example code shown in the README, this didn't work for me. I had to wrap it in a ```fn (main)```. It would be easier for people like me if the example code was a MWE, so I have modified the code and copied it below. It may be helpful to update it to help others :-)

Great job BTW... it seems to work with my USB device !!!

```
/****************************************************************************
    Copyright (c) 2018 Roland Ruckerbauer All Rights Reserved.

    This file is part of hidapi-rs, based on hidapi-rs by Osspial
****************************************************************************/

//! Opens the first hid device it can find, and reads data in a blocking fashion
//! from it in an endless loop.

extern crate hidapi;

use hidapi::{HidApi, HidError};

fn main() {
    let api = hidapi::HidApi::new().unwrap();
    // Print out information about all connected devices
    for device in api.device_list() {
        println!("{:#?}", device);
    }

    // Connect to device using its VID and PID
    let (VID, PID) = (0x0123, 0x3456);
    let device = api.open(VID, PID).unwrap();

    // Read data from device
    let mut buf = [0u8; 8];
    let res = device.read(&mut buf[..]).unwrap();
    println!("Read: {:?}", &buf[..res]);

    // Write data to device
    let buf = [0u8, 1, 2, 3, 4];
    let res = device.write(&buf).unwrap();
    println!("Wrote: {:?} byte(s)", res);
}
```