ruabmbua/hidapi-rs #166: Panic at expect "udev strings should always be utf8"

hulthe, 2025-01-09:
Hello! I'm not completely sure, but seems I have some strange device plugged in that breaks the assumption that udev strings must be utf8. I'm using probe-rs, which becomes unusable because of this.

Here's the backtrace:
```
thread 'main' panicked at /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/hidapi-2.6.3/src/linux_native.rs:386:19:
udev strings should always be utf8
stack backtrace:
   0: rust_begin_unwind
   1: core::panicking::panic_fmt
   2: hidapi::linux_native::device_to_hid_device_info
   3: <core::iter::adapters::filter::Filter<I,P> as core::iter::traits::iterator::Iterator>::next
   4: <alloc::vec::Vec<T> as alloc::vec::spec_from_iter::SpecFromIter<T,I>>::from_iter
   5: hidapi::linux_native::HidApiBackend::get_hid_device_info_vector
   6: hidapi::HidApi::add_devices
   7: hidapi::HidApi::new
   ...
```

Can provide more details if needed :)

ruabmbua, 2025-01-09:
RUST_BACKTRACE=full with debug instead of release build could be useful.

ruabmbua, 2025-01-09:
Also if you can, maybe include some debug printed output of the value that fails to convert / some info about the specific hid device, which makes it fail.

hulthe, 2025-01-10:
I found the culprit! It wasn't a physical device, it was a virtual UHID device. I turned of the service that was adding it and probe-rs works now. As for the value that fails to convert, it appears to just be random bytes that change every time I restart the UHID service. One `dbg!`-printed example: `"\xD64%\xB9\u{4}\u{1c}\xA4\xFD"`

Here's the full backtrace in case you need it:
<details><summary>Backtrace</summary>

```
thread 'main' panicked at /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/hidapi-2.6.3/src/linux_native.rs:386:19:
udev strings should always be utf8
stack backtrace:
   0:     0x5dad1f5686ca - std::backtrace_rs::backtrace::libunwind::trace::hfe0e1def394da018
                               at /rustc/9fc6b43126469e3858e2fe86cafb4f0fd5068869/library/std/src/../../backtrace/src/backtrace/libunwind.rs:116:5
   1:     0x5dad1f5686ca - std::backtrace_rs::backtrace::trace_unsynchronized::h96fbb6d117554616
                               at /rustc/9fc6b43126469e3858e2fe86cafb4f0fd5068869/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
   2:     0x5dad1f5686ca - std::sys::backtrace::_print_fmt::he1169c6f8956325b
                               at /rustc/9fc6b43126469e3858e2fe86cafb4f0fd5068869/library/std/src/sys/backtrace.rs:66:9
   3:     0x5dad1f5686ca - <std::sys::backtrace::BacktraceLock::print::DisplayBacktrace as core::fmt::Display>::fmt::ha4a311b32f6b4ad8
                               at /rustc/9fc6b43126469e3858e2fe86cafb4f0fd5068869/library/std/src/sys/backtrace.rs:39:26
   4:     0x5dad1f596213 - core::fmt::rt::Argument::fmt::h8ccd2627d2418ed3
                               at /rustc/9fc6b43126469e3858e2fe86cafb4f0fd5068869/library/core/src/fmt/rt.rs:177:76
   5:     0x5dad1f596213 - core::fmt::write::h1866771663f62b81
                               at /rustc/9fc6b43126469e3858e2fe86cafb4f0fd5068869/library/core/src/fmt/mod.rs:1189:21
   6:     0x5dad1f563623 - std::io::Write::write_fmt::hb549e7444823135e
                               at /rustc/9fc6b43126469e3858e2fe86cafb4f0fd5068869/library/std/src/io/mod.rs:1884:15
   7:     0x5dad1f568512 - std::sys::backtrace::BacktraceLock::print::hddd3a9918ce29aa7
                               at /rustc/9fc6b43126469e3858e2fe86cafb4f0fd5068869/library/std/src/sys/backtrace.rs:42:9
   8:     0x5dad1f56999d - std::panicking::default_hook::{{closure}}::h791f75256b902d7d
                               at /rustc/9fc6b43126469e3858e2fe86cafb4f0fd5068869/library/std/src/panicking.rs:268:22
   9:     0x5dad1f5697e3 - std::panicking::default_hook::h82cc572fcb0d8cd7
                               at /rustc/9fc6b43126469e3858e2fe86cafb4f0fd5068869/library/std/src/panicking.rs:295:9
  10:     0x5dad1f569f77 - std::panicking::rust_panic_with_hook::he21644cc2707f2c4
                               at /rustc/9fc6b43126469e3858e2fe86cafb4f0fd5068869/library/std/src/panicking.rs:801:13
  11:     0x5dad1f569dd6 - std::panicking::begin_panic_handler::{{closure}}::h42f7c414fed3cad9
                               at /rustc/9fc6b43126469e3858e2fe86cafb4f0fd5068869/library/std/src/panicking.rs:667:13
  12:     0x5dad1f568ba9 - std::sys::backtrace::__rust_end_short_backtrace::ha26cf5766b4e8c65
                               at /rustc/9fc6b43126469e3858e2fe86cafb4f0fd5068869/library/std/src/sys/backtrace.rs:170:18
  13:     0x5dad1f569a9c - rust_begin_unwind
                               at /rustc/9fc6b43126469e3858e2fe86cafb4f0fd5068869/library/std/src/panicking.rs:665:5
  14:     0x5dad1d45b350 - core::panicking::panic_fmt::h74866b78e934b1c0
                               at /rustc/9fc6b43126469e3858e2fe86cafb4f0fd5068869/library/core/src/panicking.rs:76:14
  15:     0x5dad1efcfc83 - hidapi::linux_native::osstring_to_string::h1618c35f0005da05
                               at /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/hidapi-2.6.3/src/linux_native.rs:386:19
  16:     0x5dad1efcd6f4 - hidapi::linux_native::device_to_hid_device_info::ha00d8c4a750250ca
                               at /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/hidapi-2.6.3/src/linux_native.rs:119:24
  17:     0x5dad1efcce75 - hidapi::linux_native::HidApiBackend::get_hid_device_info_vector::{{closure}}::h65a6bc96c84d99c3
                               at /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/hidapi-2.6.3/src/linux_native.rs:50:34
  18:     0x5dad1efd758c - core::iter::adapters::filter_map::filter_map_try_fold::{{closure}}::h0b69e243a1a19d53
                               at /home/user/.rustup/toolchains/1.84.0-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/iter/adapters/filter_map.rs:49:28
  19:     0x5dad1efdb389 - core::iter::traits::iterator::Iterator::try_fold::h25f2854a9ed793f5
                               at /home/user/.rustup/toolchains/1.84.0-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/iter/traits/iterator.rs:2370:21
  20:     0x5dad1efd5e80 - <core::iter::adapters::filter_map::FilterMap<I,F> as core::iter::traits::iterator::Iterator>::try_fold::h34470d33fd9373a9
                               at /home/user/.rustup/toolchains/1.84.0-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/iter/adapters/filter_map.rs:140:9
  21:     0x5dad1efc8f12 - <core::iter::adapters::fuse::Fuse<I> as core::iter::adapters::fuse::FuseImpl<I>>::try_fold::h9d0e5c3020a46397
[... cut ...]
                               at /rustc/9fc6b43126469e3858e2fe86cafb4f0fd5068869/library/std/src/rt.rs:174:20
  57:     0x5dad1d56734a - std::rt::lang_start::h871d8b31a66ccc25
                               at /home/user/.rustup/toolchains/1.84.0-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/rt.rs:194:17
  58:     0x5dad1d535f5e - main
  59:     0x7927fca86e08 - <unknown>
  60:     0x7927fca86ecc - __libc_start_main
  61:     0x5dad1d45bd15 - _start
  62:                0x0 - <unknown>
```

</details> 

Thanks for taking the time. Have a nice weekend!

ruabmbua, 2025-01-10:
Can you provide me with the details on the uhid service? If possible I‘d like to try reproduce the issue myself and see if it’s a bug in our code. Random bytes could be a read from uninit memory.

hulthe, 2025-01-10:
Sure, it's this one: https://github.com/tillitis/tkey-fido

Should be enough to build and run it.

stfrick, 2025-02-05:
Can confirm. The same happens with tpm-fido (https://github.com/psanford/tpm-fido)

carlosmn, 2025-12-19:
It looks like this is the serial number. I would expect any sensible device to have something that's ASCII here but a panic is not an acceptable outcome here. The hidapi version uses `wchar_to_string` which would return `WcharString::Raw` in this case as it's unable to convert it to a `String`.

While it's true we would expect strings from udev to be UTF-8, it looks like this is a misapplication or misunderstanding as these some of the fields we use here are not from udev but completely user controlled.

carlosmn, 2025-12-19:
Both of those fido virtual devices end up using the same uhid library which detects if you never provided a serial number and provides its own by filling it with random bytes https://github.com/psanford/uhid/blob/a7c1e481958d6d8861a35d61a3f0734b94848283/uhid.go#L229-L235 which I guess is fine, but it would probably be better to stick to printable characters.

An interesting way to fuzz random bits of the stack though.